mavlink/mavros #1355: RC Override strange behavior

reyes73a, 2019-11-20:
Hey guys,

So I am seeing a very strange issue when using **/mavros/rc/override** and I cannot understand what is causing this and why. When I set the value of any channel programmatically it works well, as expected, with very little latency. However, as soon as I send it (short) 65535 to give back the control to the rc transmitter it takes about 2 seconds for it to give back control. In the meantime, the value for that channel freezes up on the last value that was programmatically assigned to it. After 2s the value unfreezes and I can see the values of the rc transmitter again. This only seems to happen whenever that particular value (65535) is sent after sending other values programmatically (i.e. not the rc transmitter). I am currently check the actual channels values through:

**rostopic echo /mavros/rc/in**

The freeze I see is also showing up on the FCU because I can see it through Mission Planner. 

Any thoughts?

Thanks!!

reyes73a, 2019-11-21:
Hey! I figured it out! Turns out you have to send a (short) 0 to return back control to the rc transmitter and not (short) 65535 which instead tells the FCU to maintain the last val on that channel. After reviewing some forums and documentation I had read that 65535 was sending back control to RC, but its not true. More searching last night revealed in the documentation otherwise. So the behavior i was seeing was actually correct because I kept telling the FCU to not change the channels values hence it was freezing them whenever (short) 65535 was sent. The reason it was freezing always for 2-3secs before giving back control I suspect is likely because the FCU has a timeout and so if the value on any channel is not changing for so long it will immediately give back control to the RC. So in summary for those struggling with RC Override - a few tips:

1. currently I use APM Copter firmware on a pixhawk cube
2. Make sure that the FCU will listen to RCOverride messages:
rosrun mavros mavparam set SYSID_MYGCS (mavros must be running prior to this)
3. Values for the channels:
- Send a short primitive value ranging from 1000-2000 to control the channel just like any other RC values in microseconds
- Send 65535 to tell the FCU not to change a channel's value (will retain the last value)
- Send 0 to give back control to the rc transmitter

I hope this helps.

closed 2019-11-21 by reyes73a

vooon, 2019-11-22:
Thank you for sharing.

https://github.com/mavlink/mavros/blob/master/mavros_msgs/msg/OverrideRCIn.msg

karthikdesai, 2020-02-11:
So i'm doing the exact way you defined @reyes73a 
1. `rosrun mavros mavparam set SYSID_MYGCS 1 `
Rostopic echo of /mavros/rc/in was 
```
header: 
  seq: 21657
  stamp: 
    secs: 1581419686
    nsecs:  19577344
  frame_id: ''
rssi: 0
channels: [1500, 1500, 1000, 1500, 1800, 1000, 1000, 1800, 1000, 0, 0, 0, 0, 0, 0, 0]
```

2.  I then tried to set RC 3rd channel to 1500 by publishing this at 5 Hz
rotopic pub -r 5 /mavros/rc/override mavros_msgs/OverrideRCIn "channels: [65535, 65535, 1500, 65535, 65535, 65535, 65535, 65535]"

3. After doing so, RC3 is set at 1500 but my other RC channels get messed up (channels 9-16).
```
header: 
  seq: 22388
  stamp: 
    secs: 1581419869
    nsecs: 168073856
  frame_id: ''
rssi: 0
channels: [1500, 1500, 1500, 1500, 1800, 1000, 1000, 1800, 60750, 21961, 0, 42144, 28159, 32703, 0, 9040]
```

Did you face the same issue, or am i doing something wrong here?

karthikdesai, 2020-02-11:
Sorry i forgot to mention my setup. This was done on Arducopter SITL.
I tried the same with mavproxy by setting `rc 3 1500` on console, and it works without messing up the other channels. 
I also saw mavros_extras has a plugin mavteleop which can be used to do rc-override. Haven't come around in figuring out how to use it yet