mavlink/mavros #1356: How to properly use /mavros/setpoint_raw/target_attitude?

reyes73a, 2019-11-21:
Hey guys,

I would like to learn how to properly use **/mavros/setpoint_raw/target_attitude** messages to command the FCU to a particular orientation angle.

Right now I create  a **mavros_msgs.AttitudeTarget** and publish it to **/mavros/setpoint_raw/target_attitude**. Inside **mavros_msgs.AttitudeTarget** i specify a quaternion that defines roll,pitch and yaw. The message is publishing correctly and if I do **rostopic echo /mavros/setpoint_raw/target_attitude** i can see the values changing accordingly, but I do not see the aircraft reacting to these commands. What I am doing wrong/missing?

I got a few questions about this that could suggest why it is not working:
1. **mavros_msgs.AttitudeTarget** contains both a setOrientation and setThrust. Do i have to set the thrust always? I am not currently.
2. **mavros_msgs.AttitudeTarget** contains a few flags to ignore certain components such as **_IGNORE_ATTITUDE, IGNORE_PITCH_RATE, IGNORE_THRUST_** etc. How do I use these properly?
3. So far my tests have been arming the drone (motor spinning) while holding it down on the ground. Then move around the sticks to make sure FCU reacts to commands and then sending the **mavros_msgs.AttitudeTarget** message. I had tried in STABILIZE, ALT_HOLD and LOITER while holding it down on the ground and it is still ignoring my commands. Do I have to be in a particular mode for this to work?
4. I currently use:
**Frame: QUAD
CubeBlack
ArduCopter V3.7.0-dev** 

Will this firmware support this functionality?

Any help is greatly appreciated!
Thanks!!!

zhenfu128, 2020-09-02:
you can set **type_mask** to ignore certain components **IGNORE_ATTITUDE, IGNORE_PITCH_RATE, IGNORE_THRUST**.  According to ```uint8 IGNORE_ROLL_RATE = 1	# body_rate.x
uint8 IGNORE_PITCH_RATE = 2	# body_rate.y
uint8 IGNORE_YAW_RATE = 4	# body_rate.z
uint8 IGNORE_THRUST = 64
uint8 IGNORE_ATTITUDE = 128	# orientation field
```
you can set the last bit of type_mask to enable IGNORE_ROLL_RATE. And if you want to enable GNORE_ROLL_RATE and IGNORE_PITCH_RATE you can set the last 2 bits of type_mask to 011(binary) i.e. 3(decimal). The others are the same.

pryre, 2021-03-30:
I have not used Arducopter firmware much, but I believe your issue is that `/mavros/setpoint_raw/target_attitude` is feedback from the FCU. This tells you how the FCU is interpreting your commands. You should be sending commands to: `/mavros/setpoint_raw/attitude`.

PX4 has limits on what the firmware will accept as a valid input, otherwise it will just ignore you (off the top of my head):
- To get attitude control: `type_mask = IGNORE_ROLL_RATE | IGNORE_PITCH_RATE | IGNORE_YAW_RATE`
- To get "normal RC" control: `type_mask = IGNORE_ROLL_RATE | IGNORE_PITCH_RATE`
- To get acro control: `type_mask = IGNORE_ATTITUDE`
I believe in all cases, you need to provide a thrust input, but I haven't double checked this, and ArduCopter may be different.

vooon, 2021-03-31:
In addition - always update header.stamp, keep updates frequent.

Also you need to set GUIDED mode for APM. https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#copter-commands-in-guided-mode-set-attitude-target

joaoalves10, 2021-09-24:
@reyes73a were you able to solve this problem? I am developing a project and i need to send yaw, pitch and roll commands to a fixed wing, could you help me?

joaoalves10, 2021-09-24:
> I have not used Arducopter firmware much, but I believe your issue is that `/mavros/setpoint_raw/target_attitude` is feedback from the FCU. This tells you how the FCU is interpreting your commands. [...]

Hi, did you try attitude control in PX4 fixed wing ? i am trying for a few days with setpoint_raw/attitude command to make the aircraft turn to a specific heading with the quarterion coordinates but it does not respond. The aircraft turns to OFFBOARD mode but just mantain its previous heading

joaoalves10, 2021-09-24:
> I have not used Arducopter firmware much, but I believe your issue is that `/mavros/setpoint_raw/target_attitude` is feedback from the FCU. This tells you how the FCU is interpreting your commands. [...]

Hi @pryre , did you try attitude control in PX4 fixed wing ? i am trying for a few days with setpoint_raw/attitude command to make the aircraft turn to a specific heading with the quarterion coordinates but it does not respond. The aircraft turns to OFFBOARD mode but just mantain its previous heading

rrajnidhi, 2022-08-09:
Hi all, i was using /mavros/setpoint_raw/attitude topic to control attititude kept type_mask to 128 so that i can do attitude and thrust cntrol. I was able to do thrust control using PID but I have doubt in doing attitude control since i am trying to do global navigation along with attitude control. I am publishing global setpoints to mavros/setpoint_raw/global topic. Anyone have idea how to control attitude while doing global navigation also?
Thanks in advance

Thomas000001, 2022-09-08:
@rrajnidhi may I ask you how do you use PID to control thrust

joaoalves10, 2022-09-08:
Hi

In my project i used a PI to control thrust. I used the velocity
measurement and velocity setpoint (airspeed that i aimed to reach) to
calculate the thrust.

Thomas000001, 2022-09-08:
do you try to control drone's attitude by pitch,roll,yaw and thrust

joaoalves10, 2022-09-08:
Only fixed wing drones.

Thomas000001, 2022-09-08:
Thank you. Do you try to control quadrotor aircraft’s attitude by pitch,
roll and thrust, or have you used body rate to control attitude

joaoalves10, 2022-09-09:
I did not work with quadcopters, only with fixed wings.

Thomas000001, 2022-09-10:
I got it. Thank you very much. You help me understand the things that I
confused

joaoalves10, 2022-10-11:
Hi,

I was able to do attitude control using setpoint_raw_attitude. I used
quartenion coordinates

MatthewFehl365, 2023-07-28:
i'm having a similar issue, if I set the bitmask to 128 to ignore attitude and only use rates i get no response, only throttle works. if i add a quaternion, regardless of the value, the vehicle will constantly yaw in the SIL. Can you post a code snippet of your working result?

BBD82500, 2024-03-03:
> […](#)
Can you share the code for how you control attitude? Thank you very much!

closed 2026-11-12 by github-actions[bot]