mavlink/mavros #1358: How to use libmavconn with TCP?

bvandonselaar, 2019-11-26:
### Issue details

I want to establish a TCP connection over ethernet to send and receive Mavlink messages. I'm reading the ROS wiki, PX4 documentation and the libmavconn github repo. But I'm still having a hard time to get started with libmavconn. This #713 helped me a lot to get started, but I'm having a problem where the tcp connection doesn't get correctly disconnected i believe.  

### MAVROS version and platform

Mavros: 0.32.2
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ x] ArduPilot
[  ] PX4

### My code
```
bool MavrosHost::link(const std::string& url)
{
    try
    {
	connection = mavconn::MAVConnInterface::open_url(url, 1, (uint8_t)mavlink::common::MAV_COMPONENT::COMP_ID_USER1);

    }
    catch(mavconn::DeviceError& e)
    {
	std::cout << e.what() << std::endl;
	return false;
    }
    connection->message_received_cb = std::bind(&MavrosHost::handle_message, this, std::placeholders::_1,
						std::placeholders::_2);

    connection->port_closed_cb = std::bind(&MavrosHost::handle_closed, this);

    return true;
}

void MavrosHost::handle_closed()
{
    std::cout << "closed" << std::endl;
}

void MavrosHost::handle_message(const mavlink_message_t* message, const mavconn::Framing framing)
{
    mavlink::MsgMap map(message);
    
    switch (message->msgid)
    {
    case mavlink::common::msg::SYS_STATUS::MSG_ID:
    {
	mavlink::common::msg::SYS_STATUS s;
	s.deserialize(map);
	std::cout << s.to_yaml() << std::endl;
    }
    break;
	
    }
}
```

### Node logs
I'm sending a normal Mavlink message with fake status data for testing. This is done with the Mavlink library from a linux computer. The client connects, sends status and disconnects a couple of times for testing.

```
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    Communication (communication/communication_node)
    Scheduling (scheduling/scheduling_node)

auto-starting new master
process[master]: started with pid [30658]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7470075a-105e-11ea-966c-000c29de82e6
process[rosout-1]: started with pid [30671]
started core service [/rosout]
process[Communication-2]: started with pid [30674]
process[Scheduling-3]: started with pid [30690]
SYS_STATUS:
  onboard_control_sensors_present: 0
  onboard_control_sensors_enabled: 0
  onboard_control_sensors_health: 0
  load: 500
  voltage_battery: 11000
  current_battery: -1
  battery_remaining: -1
  drop_rate_comm: 0
  errors_comm: 0
  errors_count1: 0
  errors_count2: 0
  errors_count3: 0
  errors_count4: 0

Error:   mavconn: tcp1: receive: End of file
         at line 224 in /tmp/binarydeb/ros-kinetic-libmavconn-0.32.2/src/tcp.cpp
SYS_STATUS:
  onboard_control_sensors_present: 0
  onboard_control_sensors_enabled: 0
  onboard_control_sensors_health: 0
  load: 500
  voltage_battery: 11000
  current_battery: -1
  battery_remaining: -1
  drop_rate_comm: 0
  errors_comm: 0
  errors_count1: 0
  errors_count2: 0
  errors_count3: 0
  errors_count4: 0

Error:   mavconn: tcp2: receive: End of file
         at line 224 in /tmp/binarydeb/ros-kinetic-libmavconn-0.32.2/src/tcp.cpp
SYS_STATUS:
  onboard_control_sensors_present: 0
  onboard_control_sensors_enabled: 0
  onboard_control_sensors_health: 0
  load: 500
  voltage_battery: 11000
  current_battery: -1
  battery_remaining: -1
  drop_rate_comm: 0
  errors_comm: 0
  errors_count1: 0
  errors_count2: 0
  errors_count3: 0
  errors_count4: 0

Error:   mavconn: tcp5: receive: End of file
         at line 224 in /tmp/binarydeb/ros-kinetic-libmavconn-0.32.2/src/tcp.cpp
SYS_STATUS:
  onboard_control_sensors_present: 0
  onboard_control_sensors_enabled: 0
  onboard_control_sensors_health: 0
  load: 500
  voltage_battery: 11000
  current_battery: -1
  battery_remaining: -1
  drop_rate_comm: 0
  errors_comm: 0
  errors_count1: 0
  errors_count2: 0
  errors_count3: 0
  errors_count4: 0

Error:   mavconn: tcp6: receive: End of file
         at line 224 in /tmp/binarydeb/ros-kinetic-libmavconn-0.32.2/src/tcp.cpp
```

### Diagnostics

```
WARNING: topic [/diagnostics] does not appear to be published yet
```

### Check ID

```
WARNING: mavros/target_system_id not set. Used default value: 1
WARNING: mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages
Unhandled exception in thread started by 
sys.excepthook is missing
lost sys.stderr
Exception in thread Thread-4 (most likely raised during interpreter shutdown):Exception in thread Thread-2 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
Traceback (most recent call last):

  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/registration.py", line 275, in start  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner

  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/timer.py", line 235, in run  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/registration.py", line 307, in run

  File "/usr/lib/python2.7/threading.py", line 203, in release  File "/opt/ros/kinetic/lib/mavros/checkid", line 90, in timer_cb

  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/core.py", line 493, in signal_shutdown<type 'exceptions.TypeError'>: 'NoneType' object is not callable

  File "/usr/lib/python2.7/threading.py", line 946, in join
<type 'exceptions.TypeError'>: 'NoneType' object is not callable
```

vooon, 2019-11-26:
What URL used? Looks like other end closes the connection.

In case if both sides it libmavconn, one side should be server (tcp-l) and other client (tcp).

bvandonselaar, 2019-11-27:
> What URL used? Looks like other end closes the connection.
> 
> In case if both sides it libmavconn, one side should be server (tcp-l) and other client (tcp).

The client is not using libmavconn right now. Just Boost Asio with the Mavlink library. I have checked how MAVConnTCPClient closes the connection
```
socket.shutdown(boost::asio::ip::tcp::socket::shutdown_send);
socket.cancel();
socket.close();
```
My client does shutdown_both, maybe that is my problem? Or should the server close the client? or the client close the server?

bvandonselaar, 2019-12-02:
I also tried it with the MAVConnTCPClient like this:
```
#include <iostream>
#include <mavconn/interface.h>

int main(void)
{
    mavconn::MAVConnInterface::Ptr client;
    try
    {
	client = mavconn::MAVConnInterface::open_url("tcp://localhost:8888");
    }
    catch(mavconn::DeviceError& e)
    {
	std::cout << e.what() << std::endl;
	return -1;
    }

    client->close();
    return 0;
}
```
but it gave me the same results with reconnecting to the server.

closed 2026-11-12 by github-actions[bot]