mavlink/mavros #1363: Running precision landing on the FCU with landing target plugin doesn't seem to be working

canberkgurel, 2019-12-11:
----

### Issue details

Hi all, 

I am working on a precision landing problem where I have a custom vision node ([aruco_detect](https://github.com/UbiquityRobotics/fiducials)) that runs on the companion computer (Ubuntu 16.04, ROS Kinetic) and passes the pose estimation to the [landing_target.cpp plugin](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/landing_target.cpp). The message type that I'm using is PoseStamped ([geometry_msgs::PoseStamped](http://docs.ros.org/melodic/api/geometry_msgs/html/msg/PoseStamped.html)) msg. I added a few print statements into landing_target.cpp and captured a snapshot of the data that the plugin is receiving. 

```
[DEBUG] Inside the pose_cb function. 
Header.stamp = 9300.447500000
Pose = position: 
  x: 1.2662
  y: -0.464283
  z: -2.76518
orientation: 
  x: -0.0109453
  y: 0.941741
  z: -0.335588
  w: -0.0195984

[DEBUG] Inside mavros::send_landing_target function.
land target type is VISION_FIDUCIAL
pos.x = -0.464283, 
angle.x = 1.15295e+18, 
pos.y = -0.464283, 
angle.y = -2.28187e-41

[DEBUG] Input to mavros::landing_target function: 
angle =  1.15295e+18, -2.28187e-41, 
distance = 3.07654, 
size_rad = 0.00186821, 0.00186821
```

When I'm testing, the drone arms, switches to the GUIDED mode, and takes off to 3m, then it flies over the aruco marker. I monitor the output of the vision node and when it starts publishing the pose estimation (a marker is being detected), the drone switches to the land mode and lands. I expect the drone to follow the landing target that is set while it's descending, which doesn't appear to be the case. It looks like it doesn't take corrective actions and instead it lands where it is.  

I suspect a few possible causes:
**(1)** I don't see [Landing Target (#149)](https://mavlink.io/en/messages/common.html#LANDING_TARGET) in MAVlink Inspector. So the data may not be making its way all the way to the Mavproxy and to simulated FCU. How can I monitor for the landing target mavlink message?
**(2)** The units of the position that's fed to the landing_target plugin is in meters, is it supposed to be number of pixels? 
**(3)** The landing_target settings in [apm_config.yaml](https://github.com/mavlink/mavros/blob/master/mavros/launch/apm_config.yaml) may not be set correctly? (see appendix B)
**(4)** The AUTO.LAND may not be the right flight mode for the drone to follow the landing targets passed to the landing_target.cpp?

Appendix A: Details of the model I'm simulating:
```
# Stack: ArduPilot
# Vehicle: Multi-Rotor
# Version: 3.6.11 
# Git Revision: f0d5929
```

Appendix B: Content of the apm_config.yaml file related to the landing target plugin.
```
# landing_target
landing_target:
  listen_lt: false
  mav_frame: "LOCAL_NED"
  land_target_type: "VISION_FIDUCIAL"
  image:
    width: 640            # [pixels]
    height: 480
  camera:
    fov_x: 2.0071286398   # default: 115 [degrees]
    fov_y: 2.0071286398
  tf:
    send: true
    listen: false
    frame_id: "landing_target"
    child_frame_id: "camera_center"
    rate_limit: 10.0
  target_size: {x:  0.3, y:  0.3}
```

Appendix C: How I read & load the apm_config.yaml file in the launch file. 
`<rosparam file="$(find mavros)/launch/apm_config.yaml" command="load" />`

Appendix D: Content of the gazebo-iris.param file.
```
# Iris is X frame
FRAME_CLASS	1
FRAME_TYPE	1
# IRLOCK FEATURE
CH8_OPT 39
PLND_ENABLED    1
PLND_TYPE       1
# SONAR FOR IRLOCK
SIM_SONAR_SCALE 10
RNGFND_TYPE 1
RNGFND_SCALING 10
RNGFND_PIN 0
RNGFND_MAX_CM 5000
```

Appendix E: Configured PLND type:
`PLND_TYPE = CompanionComputer`

**References**:
https://github.com/UbiquityRobotics/fiducials
https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/landing_target.cpp
http://docs.ros.org/melodic/api/geometry_msgs/html/msg/PoseStamped.html
https://github.com/mavlink/mavros/blob/master/mavros/launch/apm_config.yaml
https://mavlink.io/en/messages/common.html#LANDING_TARGET

Adding my colleagues to the thread: @khabbab-nazar @m1baldwin

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: ?3.7.1?

aydal, 2019-12-29:
@canberkgurel Were you able to solve it?

canberkgurel, 2019-12-30:
@aydal No. But I have a few updates.

I started using the [cartesian_to_displacement](https://github.com/mavlink/mavros/blob/63183a6fee196b0620a7f7749c5ddeae5b50c669/mavros_extras/src/plugins/landing_target.cpp#L260) function in void send_landing_target. As a result the drone started oscillating when I trigger landing. This answers my Q1: the data flow is fine. 

As you can see in the GIF below, the drone starts circling above the target and the eventually fades away from the target. I still need help to solve this problem.

[image]

aydal, 2019-12-31:
Hello @canberkgurel

khbnzr, 2019-12-31:
@aydal We are using the APM software stack and in that the mode for precision landing is "land"(after setting the proper precland enable params).

mohammedshafeeqet, 2020-06-22:
@canberkgurel  @khabbab-nazar @m1baldwin
Were you able to see the messages in MAVlink inspector?

canberkgurel, 2021-08-28:
@DoneMan No, not at the time. We'll re-attempt soon with more recent mavros, mavlink, and PX4-Autopilot versions.

BrenoPimentel, 2022-08-16:
Did you guys solve it?

closed 2026-11-12 by github-actions[bot]