mavlink/mavros #1364: Mavros - Gazebo and Px4 HITL not working - Param timeouts

NicoMandel, 2019-12-12:
# Issue details

When trying to start the px4 hardware with mavros and HITL, I am unable to start simulations because the flight controller gets either stuck in a loop of:
**Case 1**
```
[Wrn] [gazebo_mavlink_interface.cpp:588] Tx queue overflow
```
**OR**
**Case 2**
```
[ WARN] [1576125196.394880331, 30.440000000]: PR: request param #8 timeout, retries left 1, and 533 params still missing
```
And I cannot run any offboard control or any other ROS nodes to send commands for mavros to forward to execute. In the beginning I also get a lot of `TM: RTT too high for timesync: ` errors and have no clue whether these are related.
Any help would be greatly appreciated!

### Proprietary changes:
*  px4 config - timesync rate disabled, [here](https://github.com/mavlink/mavros/blob/563f7f5e12e886473019fa80521b9e735298a747/mavros/launch/px4_config.yaml#L12)
* Copied a new iris_hitl.sdf file as per recommendations of Px4 HITL - [including suggested changes](https://dev.px4.io/v1.9.0/en/simulation/hitl.html#simulator-setup)
* Changed [mavros_posix_sitl.launch](https://github.com/PX4/Firmware/blob/master/launch/mavros_posix_sitl.launch) to include, to enable naming the model `iris` and launching the proprietary `iris_hitl.sdf` file mentioned above (being called through `iris_bottom_cam` in the launch below)
```
<arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="model_name" default="iris_bottom_cam"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg model_name)/$(arg model_name).sdf"/>
```
* Launch file to execute the code:
```
<?xml version="1.0"?>
<launch>
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <arg name="gui" default="true"/>
    <arg name="vehicle" default="iris"/>
    <arg name="verbose" default="true"/>
    <arg name="world" default="$(find aruco_client)/worlds/word2vec_empty.world"/>
    <arg name="hitl" default="true"/>

    <arg name="fcu_url" default="/dev/ttyACM1:57600"/> <!--serial://-->
    <arg name="model_name" default="iris_bottom_cam"/>
    <arg unless="$(arg hitl)" name="model_name_hitl" value="$(arg model_name)"/>
    <arg if="$(arg hitl)" name="model_name_hitl" value="$(arg model_name)_hitl"/>

    <!-- 1. Including the launch file for the Mavros -->
    <include file="$(find px4)/launch/mavros_posix_sitl.launch">
        <arg name="x" value="$(arg x)"/>
        <arg name="y" value="$(arg y)"/>
        <arg name="z" value="$(arg z)"/>
        <arg name="R" value="$(arg R)"/>
        <arg name="P" value="$(arg P)"/>
        <arg name="Y" value="$(arg Y)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="vehicle" value="$(arg vehicle)" />
        <arg name="verbose" value="$(arg verbose)" />
        <arg name="world" value="$(arg world)" />
        <arg name="model_name" value="$(arg model_name_hitl)" />
        <arg name="fcu_url" value="$(arg fcu_url)"/>
    </include>
</launch>
```

### MAVROS version and platform

Mavros: 0.33.0
ROS: Melodic
Ubuntu: 18.04

[... 520 lines cut ...]
    values: 
      - 
        key: "Voltage"
        value: "65.54"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "-1.0"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 143 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 74, 77, 111, 147, 340, 245, 22, 36

```

dayjaby, 2020-05-20:
Any updates on this?

NicoMandel, 2020-05-20:
Hey David. Yeah I had issues concerning the `.sdf` file which loads the model and the `fcu_url` in my mavros node.
I modified the `iris.sdf` according to the specifications in the instructions from [Px4](https://dev.px4.io/v1.10/en/simulation/hitl.html) and then rewrote the `mavros fcu_url` to also point to the dev port in the [launch file](https://github.com/PX4/Firmware/blob/504794eddcd91f263431901e4b86c684987d4743/launch/mavros_posix_sitl.launch#L25). However, this causes a hardware lock issue because gazebo (through the sdf) as well as mavros are trying to access the same hardware port.
You should have a look what the "SDK" udp port in the `iris.sdf` file says, in my case that is `<sdk_udp_port>14540</sdk_udp_port>`, which is what the fcu_url should attach to. For me that was: 
` <arg name="fcu_url" default="udp://:14540@localhost:14557"/>`
And then it worked. I amusing PX4 1.10 though and a Pixracer. With the pixracer, I have to give it a little while to start up before I can run the HITL after attaching it to the USB. I also have to replug every time before restarting due to a GPS lock.
Hope this helps.
Viel Glueck

dayjaby, 2020-05-21:
And the Tx queue overflow disappeared? Struggling with this overflow + no vehicle_local_position and vehicle_global_position, even though it shows 10 satellites.

NicoMandel, 2020-05-21:
So my colleague helped me fix this and applying these fixes. Yes, because tx queue overflow is caused by the serial port not being able to be opened, because another resource is trying to access it. When using the `fcu_url` with a serial port `/dev`, as well as specifying the serial port in the `.sdf` file which is loaded during hitl, two competing resources are trying to access the same hardware serial port, which is [not really possible](https://learning.oreilly.com/library/view/linux-network-administrators/1565924002/ch04s03.html) in linux.
To bypass this, the `sdk_udp` port is specified in the hitl `.sdf` file. This allows the information being passed between gazebo and the hardware connected via serial to be accessed with the udp protocol. The information is kind-of multiplexed, because there is also a GCS udp port specified, if you look closely.
When specifying the `fcu_url` in the udp format. (i think it is udp://:@in:host:out or something), this allows a process to lock to that udp port and collect that information. The network setup is shown in the [ROS Interface](https://dev.px4.io/v1.9.0/en/simulation/ros_interface.html) part of the px4 documentation. I struggled with this for about a month before my colleague could help me fix this.

I have not yet figured out how to connect a onboard computer to the host over the network, run mavros on this and run a hitl-simulation on the host. I think there would be some complex tcp/udp tunneling happening, which I could not figure out.
I hope this helps
Cheers

NicoMandel, 2020-05-21:
Also make sure you activated the serial ports and set the baudrate right in the parameters in the GCS.

closed 2026-11-12 by github-actions[bot]