mavlink/mavros #1365: Difference between the two accelerations : mavros/setpoint_accel/accel and 'mavros/setpoint_raw/local', PositionTarget.msg

Jaroan, 2019-12-12:
I am trying to implement acceleration control on an Iris Quadrotor using PX4 SITL on ROS Gazebo 9 using Mavros. I want to use a body frame for getting the positions and velocities of the quadrotor and feed in the accelerations to the quad after using PD Control. I wanted to know the difference between the two topics that can be used to publish accelerations namely: mavros/setpoint_accel/accel and 'mavros/setpoint_raw/local', PositionTarget.msg (geometry_msgs/Vector3 acceleration_or_force).

Jaeyoung-Lim, 2019-12-12:
@Jaroan Acceler

canberkgurel, 2019-12-12:
Setpoint topics are inputs to mavros and the other ones are the outputs. For instance, you can assign acceleration setpoint for the drone to converge, and then you can monitor the acceleration of the drone in real-time from the mavros/local_position/accel topic. I hope this helps.

Jaroan, 2019-12-12:
Thank you very much for this info. Also, what is the difference betweensetpoint_raw and setpoint_accel?

Jaeyoung-Lim, 2019-12-14:
@Jaroan It uses the same mavlink message. It just uses a different plugin in mavros

Jaroan, 2019-12-16:
Oh alright, thank you for this info!

Jaroan, 2019-12-16:
Update: I have used the following topics for position, velocity and acceleration controls:
```
#Subscribe to drone's local position
rospy.Subscriber('mavros/local_position/pose', PoseStamped, cnt.posCb)

#Subscribe to drone's velocity
rospy.Subscriber("/mavros/local_position/velocity", TwistStamped , cnt.velCb) 

#Setpoint publisher
sp_pub = rospy.Publisher('mavros/setpoint_raw/local', PositionTarget, queue_size=1)
```
 The command
`rostopic echo /mavros/setpoint_raw/local` shows that the required values are given to the quad as shown in the left terminal of the image:
[image]
```
---
header: 
  seq: 1620
  stamp: 
  secs: 0
    nsecs:         0
  frame_id: ''
coordinate_frame: 1
type_mask: 1080
position: 
  x: 0.0
  y: 0.0
  z: 4.0
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration_or_force: 
  x: 1.60403175801
  y: 1.69204383269
  z: 0.0
yaw: 0.0
yaw_rate: 0.0
```
but the drone doesn't move from it's fixed altitude position. Just hovers around that point after giving a fixed set velocity and controlled accelerations.

Jaroan, 2019-12-17:
Update: I’ve tried three topics to publish the accelerations into:

- /mavros/local_position/accel [geometry_msgs/AccelWithCovarianceStamped] , 

- mavros/setpoint_accel/accel [geometry_msgs/Vector3Stamped] ,

- /mavros/setpoint_raw/local [mavros_msgs/PositionTarget] (geometry_msgs/Vector3 acceleration_or_force)

The first and third caused no motion to the quad even though they published into the topics well (checked using ‘rostopic echo’). The second topic gave an initial acceleration in the x direction and quickly the drone crashed at a distance. After discussions, I am guessing it is related to the discussions in this issue https://github.com/PX4/Firmware/pull/4010#issuecomment-403518185

RayhanHaqi, 2020-09-08:
Maybe u can try publish to mavros/setpoint_position/local, it worked for me.

Jaeyoung-Lim, 2020-09-08:
@Jaroan Offboard acceleration setpoints are not something that works in PX4 yet

closed 2020-09-08 by Jaroan

Wenlong0605, 2020-12-01:
> @Jaroan Offboard acceleration setpoints are not something that works in PX4 yet

So what can we do if we want to use accelerations as our controller's outputs. Should we use attitude setpoint instead?

shubham-shahh, 2021-11-09:
> @Jaroan Offboard acceleration setpoints are not something that works in PX4 yet

hey, How can I get the current acceleration of the copter? @Jaeyoung-Lim 

thanks

Jaeyoung-Lim, 2021-11-09:
@shubham-shahh You can check which topic contains acceleration: http://wiki.ros.org/mavros

shubham-shahh, 2021-11-09:
> @shubham-shahh You can check which topic contains acceleration in the mavros wiki: http://wiki.ros.org/mavros

I tried the following topic mavros/setpoint_accel/accel but the topic is empty

javiAldazabal, 2021-12-13:
@Jaeyoung-Lim Then it is not possible, at the moment, sending acceleration values using /mavros/setpoint_raw/local? And we have to use /mavros/setpoint_accel/accel? Maybe this is related with [this](https://discuss.px4.io/t/unable-to-set-accels-in-offboard-using-setpoint-raw-local/25067) issue. Thanks!