mavlink/mavros #1367: Unable to connect Pixhawk with Auvidea J120

Deepakgthomas, 2019-12-14:
Hello,

As per the APM website, I set up ROS and used it to connect the Pixhawk with my Jetson TX2 (using Auvidea J120 carrier board) - 
`roslaunch mavros apm.launch fcu_url:=udp://:1500@
`
`nvidia@nvidia:~$ roslaunch mavros apm.launch fcu_url:=udp://:1500@
... logging to /home/nvidia/.ros/log/8c4f81a8-1ebd-11ea-815b-00044ba82086/roslaunch-nvidia-12446.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia:41293/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:1500@
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: base_link
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
[... 166 lines cut ...]
[ INFO] [1576367681.812500517]: Plugin trajectory initialized
[ INFO] [1576367681.813156494]: Plugin vfr_hud loaded
[ INFO] [1576367681.816535617]: Plugin vfr_hud initialized
[ INFO] [1576367681.816758437]: Plugin vibration blacklisted
[ INFO] [1576367681.817270892]: Plugin vision_pose_estimate loaded
[ INFO] [1576367681.844038238]: Plugin vision_pose_estimate initialized
[ INFO] [1576367681.844252481]: Plugin vision_speed_estimate blacklisted
[ INFO] [1576367681.844757673]: Plugin waypoint loaded
[ INFO] [1576367681.860600407]: Plugin waypoint initialized
[ INFO] [1576367681.860847898]: Plugin wheel_odometry blacklisted
[ INFO] [1576367681.861411427]: Plugin wind_estimation loaded
[ INFO] [1576367681.865396607]: Plugin wind_estimation initialized
[ INFO] [1576367681.865706147]: Built-in SIMD instructions: ARM NEON
[ INFO] [1576367681.865898310]: Built-in MAVLink package version: 2019.11.11
[ INFO] [1576367681.866098889]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1576367681.866278412]: MAVROS started. MY ID 1.240, TARGET ID 1.1
`
However, I'm unable to communicate with my Pixhawk at all. For instance, when I try to arm it, here is what I get - 

```
rosrun mavros mavsafety arm
Request failed. Check mavros logs
```

I have checked all previous posts that faced similar issues but their suggestions hasn't resolved my issue.

khancyr, 2019-12-16:
hello,

You miss read the instruction from https://ardupilot.org/dev/docs/ros-connecting.html . the line `roslaunch mavros apm.launch fcu_url:=udp://:14855@` is to connect mavros to a TX2 with APsync ...
to connected to a board on serial use :
`roslaunch mavros apm.launch fcu_url:=serial:///path/to/serial/device[:baudrate]` as stated in mavros instruction

Deepakgthomas, 2019-12-17:
Thank you @khancyr Can you also please tell me how to find the value for, '///path/to/serial/device'?

khancyr, 2019-12-17:
it depends on your board and OS, please look for it in your board documentation. On debian type linux like ubuntu, it will be in /dev/ttyXX (XX could be anything e.g. S1, USB0, 1 , etc.). but that is better to read your documentation

cacaroxy, 2020-02-26:
Write it as below.
If u are connected to UART

fcu_url:="/dev/ttyTHS1:921600"

and connect to telemetry port 

in case J120

closed 2026-11-11 by github-actions[bot]