mavlink/mavros #1368: How to import mavros that is built from source into my package

HenryHuYu, 2019-12-20:
### Issue details

I use Jetson nano and successfully built mavros in catkin_ws. I can run roslaunch mavros px4.launch now. 

but since the installation process uses wstool, as I know, it's not a good idea to use catkin_make and catkin build at the same time. So, I create another catkin_ws_another to develop my project. but in my project, I got this error: cannot find mavrosConfig.cmake. 

I guess that it is probably because the mavrosConfig.cmake is in ~/catkin_ws/devel/mavros/share instead of binary installation (/opt/ros/melodic/share)

So how do I fix this? move the file or fix Cmakelists?

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Melodic
Ubuntu: 18.04

vooon, 2019-12-22:
In the first place why you want to use catkin_make at all?

HenryHuYu, 2019-12-23:
> In the first place why you want to use catkin_make at all?

do you mean that I should not use catkin_make? Then which command should I use? Sorry 
I am new in ROS program

vooon, 2019-12-23:
https://github.com/mavlink/mavros/tree/master/mavros#source-installation

HenryHuYu, 2020-01-10:
@vooon Hi, I am using wstools now. but  still got that problem. If I install mavros from source, how to import it into another catkin_workspace?

TSC21, 2020-01-10:
> @vooon Hi, I am using wstools now. but still got that problem. If I install mavros from source, how to import it into another catkin_workspace?

That's part of ROS tutorials :)

http://wiki.ros.org/catkin/Tutorials/workspace_overlaying

closed 2026-11-11 by github-actions[bot]