mavlink/mavros #1369: Release 1.0.0

vooon, 2019-12-30:
Hello all,

While i can't say that mavros is stable enough, but it's time to move to the new ROS.
So current code now should be kept as "old stable code that we use for a while".

ROS2 have a lots of differences, so next iteration of mavros would be quite different from that code.
It also opens us a window to redesign to achieve long-lived feature requests like gcs plugins and swarm support (but i can't guarantee that).

So:
1. Upcoming 0.34 will be released as 1.0.0
2. From version 2.0.0 we start new code for ROS2

vooon, 2020-01-01:
Kinetic: https://github.com/ros/rosdistro/pull/23391
Melodic:

closed 2020-01-01 by vooon

vooon, 2021-05-28:
Foxy alpha release: https://github.com/ros/rosdistro/pull/29704

dzywater, 2021-09-14:
Hi, @vooon, see https://px4.io/wp-content/uploads/2019/06/Presentation-ROS2-powered-PX4.pdf  ： page23~page24
[image]
[image]

I want to know whether mavros2 still has above disadvantages.

vooon, 2021-09-14:
@dzywater with mavros 2 you'll have most of the same APIs as you have on ROS1.
So perhaps for parties who's already uses ROS1 that'll be easier option.

If we're talking about uORB vs Mavlink, then there is not difference. MAVROS v2 still talks to Mavlink, so on PX4 you'll don't get access to some of internals. But on other hand, you have an option to work with Ardupilot, which do not have uORB at all.

TSC21, 2021-09-14:
@dzywater that presentation is from 2019 and by then that was the current state. But, although you can use ROS2 to interface with PX4, the main advantage is that you don't require MAVLink anymore to interface with the flight controller internals. And since in the end you would always endup interfacing with a pub/sub infrastructure similar to ROS (which is the case of uORB), you don't need MAVLink for nothing.
Adding up that we have overused what MAVLink was supposed to be for, which is not for sure a Pub/Sub infrastructure for communication between companion computers and PX4-enabled flight controllers. For this same purpose we have the microRTPS/ROS2 bridge, offering a native interface between ROS2 topics and the PX4 internals, without having to add even another protocol on top (MAVLink). And soon supporting the DDS-XRCE protocol.

dzywater, 2021-09-18:
> @dzywater that presentation is from 2019 and by then that was the current state. But, although you can now use mavros to interface ROS2 with PX4, the main advantage is that you don't require MAVLink anymore to interface with the flight controller internals. And since in the end you would always endup interfacing with a pub/sub infrastructure similar to ROS (which is the case of uORB), you don't need MAVLink for nothing.
> 
> Adding up that we have overused what MAVLink was supposed to be for, which is not for sure a Pub/Sub infrastructure for communication between companion computers and PX4-enabled flight controllers. For this same purpose we have the microRTPS/ROS2 bridge, offering a native interface between ROS2 topics and the PX4 internals, without having to add even another protocol on top (MAVLink). And soon supporting the DDS-XRCE protocol.
> 
> The current recommendation for the development and maintainance team of PX4 for interfacing ROS2 with PX4 is through the microRTPS bridge.

@TSC21 Could ROS2 or MicroRTPS replace mavlink completely?   Where ROS2 cannot ? Or Where is the mavlink better than ROS2?  I feel that mavlink can do where ROS2 can do also.

vooon, 2021-09-26:
@dzywater don't know. Perhaps similar as link should contribute most of the delay.