mavlink/mavros #1375: Waypoint push low resolution

jimovonz, 2020-01-13:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Setting waypoints looses resolution with lat/long accurate only to approx 5 decimal places. This results in waypoints that are up to around +/1 1m out.

### MAVROS version and platform

Mavros: 1.0.0
ROS: Melodic
Ubuntu: ?18.04?

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2020-01-13:
Mavlink protocol uses float32 for coordinates.

LorenzMeier, 2020-01-13:
Most implementations have switched to scaled integers, which do not introduce a loss of precision. MAVROS should do the same.

jimovonz, 2020-01-13:
Technically perhaps you could argue that it is not a bug but why limit WP precision to +/- 1m unnecessarily? With the proliferation of consumer RTK GNSS, this will become a serious bottleneck for precision navigation....
The same problem was recently addressed in Mavproxy. Wp pull also suffers from the same issue.

vooon, 2020-01-14:
@LorenzMeier waypoint plugin need huge revision to support MISSION_ITEM_INT and to support mission_type extension.
And I'm thinking to spend my limited free time to migrate to ROS2.
But of course PR welcome.

bromeara, 2020-01-21:
@vooon from the ROS side how should this problem be approached in terms of interface. Would you want to have the option available to use MISSION_ITEM or MISSION_ITEM_INT(I would think through the use of a flag in the message) or would it be better to do the change silently behind the scenes and keep the messages the same?

vooon, 2020-01-21:
@bromeara it is possible to do param for: auto, mission_item, mission_item_int

`sys_status` gets autopilot capabilities and it should be available trough UAS.
So plugin may check the state before start of sequence of requests and choose old/new message set to operate.

bromeara, 2020-01-21:
@vooon when you mentioned the mission_type extension, was this in reference to adding support for the parameter in the ROS message or is there something that needs to be changed in the underlying mavlink implementation to add this Mavlink 2 extension? It seems like you already use it with the current waypoint_item implementation

vooon, 2020-01-21:
@bromeara if you mean libmavconn, then no, it is already support v2. Just now mission_type is hardcoded to only one supported table - missions. While it is possible to setup more tables, but that require changes in the plugin (to support several mission lists). That would be nice to have, but probably that is another topic, because that would also bread ROS API (messages).

bromeara, 2020-01-21:
@vooon Makes sense, would it make sense for fence and rally points to be different plugins to maintain that API? I haven't written a plugin for mavros before, is it possible to have different plugins handle the same message and then throw it away if the fence plugin received a mission mission_type MISSION_ITEM? Something like this could let the waypoint interface stay the same.

bromeara, 2020-01-21:
So after looking at the APM firmware (which is what I work with), it looks like they don't have the ability to send a MISSION_REQUEST_INT message but they just treat MISSION_ITEM and MISSION_ITEM_INT messages identically, by converting MISSION_ITEM into MISSION_ITEM_INT. This isn't too much of an issue if I use the parameter approach for deciding which MISSION_ITEM message to use but I'm not sure if other Mavlink conformant autopilot stacks would have the MISSION_REQUEST_INT message, and perform more closely to the protocol defined [here](https://mavlink.io/en/services/mission.html#uploading_mission). Can someone that works with the px4 or other stacks see if their firmware works in a similar way?

@LorenzMeier How do other implementations select between sending a MISSION_ITEM_INT and a MISSION_ITEM if which one is used is actually dictated by the response of the flight controller after a call of MISSION_COUNT?

bromeara, 2020-01-22:
I need to have a different int encoding factor depending on whether the frame is local or global. However, there is one frame (MAV_FRAME_MISSION) that I am not sure how it is used. In the ardupilot docs I saw [MAV_CMD_DO_DIGICAM_CONFIGURE](https://ardupilot.org/copter/docs/common-mavlink-mission-command-messages-mav_cmd.html#mav-cmd-do-digicam-configure) uses param7, traditionally altitude so not effected but its more of an example, would this be the type of mission_item that would use a MAV_FRAME of 2? I set the encoder factor of MAV_FRAME 2 to be 1 so any values with a MAV_FRAME of 2 would be unencoded but is this what MAV_FRAME 2 is used for or should I actually be encoding the values?

vooon, 2020-01-22:
@bromeara the problem with different plugin is that it would be copy-paste of wp with some minor changes, and i do not like to violate DRY rule. Yes, it is possible to move some parts into the mixin, but, probably, it would be difficult to support. So i think that easiest way - to extend wp. But, as i said, support for different mission types not necessary to do from the beginning.

bromeara, 2020-01-23:
I've finished replacing that entirely replaces MISSION_ITEM with MISSION_ITEM_INT and will be working on the code that allows for backwards compatibility. @vooon what is your goal in terms of implementation support? Do you just want me to confirm it works for px4 and arducopter or are there others I should verify it works with? Due to the fact that Arducopter can't send MISSION_REQUEST_INT I will have to send MISSION_ITEM_INT in response to MISSION_REQUEST messages for ardupilot specifically and trust the MAV_PROTOCOL_CAPABILITY mask. But if px4 or another mavlink device support MISSION_INT in MAV_PROTOCOL_CAPABILITY but can send both MISSION_REQUEST and MISSION_REQUEST_INT I probably shouldn't send a MISSION_REQUEST_INT to a MISSION_REQUEST like I would for ardupilot.

vooon, 2020-01-23:
@bromeara yes, PX4 and APM is enough. And yes, check for API quirk is fine, just leave a note in module comments about quirks, because it's easy to forget. Also you may leave a link to this issue.

bromeara, 2020-02-21:
@vooon What cpp standard is used to build this package?

bromeara, 2020-04-20:
Hi @vooon, A few other priorities came up but I am revisiting this issue, one question I have about what would be considered backward compatible is that the current WaypointItem class has the members of x_lat, y_long, and z_alt which are doubles. It seems these are solely to store the information coming from the .msg interface and maintain full precision. However, since the mavlink protocol will fill these doubles with floats anyways are these necessary to maintain? I will also be adding comments to the code alerting that the code packs float and if greater accuracy is needed to use the int version. These variables seem like an extra layer of indirection that is not really necessary.

vooon, 2020-04-21:
@bromeara as i remember, i added doubles to keep full precision to be able to make integer version of the message transparently. And also to have less issues with other ros messages, as they commonly use doubles.

bromeara, 2020-04-21:
@vooon Yeah I understood it from a ROS message side but I was talking about from the class side and realized that it's dependent on how I am reworking things so we can review it when I submit the merge request, thank you. 

Also, I am increasing code reuse by using boost::variant to allow for code reuse and it might be hard to say 100% without seeing the code but does this raise any red flags for you? Other forms of polymorphism didn't really cut it but I'm sure what the general policy on external libraries is.

PR #1415 (2020-05-22): Mission item int
This pull request adds a class that inherits from MISSION_ITEM_INT and allows for the usage of it with the parameter use_mission_item_int if it's not given it will use the flight controller capabilities. I used cog where it felt practical, I used boost variant to improve code reuse slightly. I am curious if I should have used cog for handle_mission_item and handle_mission_item_int since they are very similar.
I switched the send_waypoint and waypoint vectors to storing the ROS messages instead of the mavlink messages so that they can be converted to float or int at will withou [... cut]
closed 2020-05-22 by vooon