mavlink/mavros #1378: Changing to GUIDED mode works only from RC remote.

ilisparrow, 2020-01-25:
----

### Issue details

Hello, 
thank you for taking the time to read my request.
My team and I have had a hard time changing the mode of the pixhawk. 
It works when we do it through the RC command (switching from manual to Guided to Auto).
But we didn't have any luck through mavros.
We tried with the code : 
```
self.cmd = rospy.ServiceProxy('/mavros/cmd/arming', CommandInt)
rospy.init_node('test_mode', anonymous=True)
self.resultat = self.mode(0,'GUIDED')
```
But we get : 
MODE: Unsupported FCU
By command line : 
```
rosservice call /mavros/set_mode "base_mode: 0 custom_mode: 'Guided'"
mode_sent: False
```
With rosrun : 
```
rosrun mavros mavsys mode -c GUIDED 
 Request failed. Check mavros logs
```
It also says for both of them:
MODE: Unsupported FCU
or by mavros command : 
they either return : 
or : 
Thank you very much in advance for your help.

### MAVROS version and platform

Mavros: Not sure, but the latest
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ x ] ArduPilot
[   ] PX4

Version: 4.0

### Node logs

```
... logging to /home/ubuntu/.ros/log/f86feb36-3f86-11ea-b655-b827ebed45c4/roslaunch-ubuntu-14610.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:40545/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
[... 434 lines cut ...]
        key: "Current"
        value: "-0.0"
      -
        key: "Remaining"
        value: "-1.0"
---

```

### Check ID

```

OK. I got messages from 1:1.

---
Received 1252 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 150, 193, 152, 22, 24, 27, 29, 30, 32, 33, 34, 163, 36, 165, 42, 178, 182, 65, 74, 111, 241, 116, 125

```

As said before, it works from the RC remote. 
We would be gratful if you could help us.
thank you very much in advance.

rsisto, 2020-05-07:
Hi, did you find a solution to this?

I'm receiving the same "Unsupported FCU" message.

Thanks!

ilisparrow, 2020-05-07:
Hello, Yes we did, 
It was because we activated the boat mode in the setting (I forgot what the exact setting was, wire-frame number or something like this) as soon as we changed it back to a land vehicle it worked again !
I hope this will help you !
Best regards.

closed 2020-06-04 by ilisparrow