mavlink/mavros #1379: /mavros/state fields meaning

elgarbe, 2020-01-28:
Hi, where do I can information about 
bool connected
bool armed
bool guided
bool manual_input
fields from /mavros/state message?

I'm switching to GUIDED mode with my Radio Control, /mavros/state->mode change to GUIDED and stay there, but my hexa do not accept /mavros/setpoint_velocity/cmd_vel commands, it stay in their position (previous flight mode was POSHOLD). I need to know how to confirm that hexa (ARducopter 3.5) is in GUIDED mode other than checking /mavros/state->mode. is /mavros/mode->guided flag field the one that I should look at? how does this flag works?

Thank!

vooon, 2020-01-29:
Guided flag just decodes "base mode" field of the [HEARTBEAT][1], so it depends on ap firmware.
Make sure that you generate stream of cmd_vel with changing stamp.

[1]: https://mavlink.io/en/messages/common.html#HEARTBEAT

elgarbe, 2020-02-02:
@vooon thank for your answer. The problem was with hosts configuration in laptop and raspberry. But AP is different to PX4. AP accept GUIDED mode with just GPS signal, no matter cmd_vel command is sended or not. PX4 reject OFFBOARD mode if not detect command messages.

closed 2020-02-02 by elgarbe