mavlink/mavros #1383: RC_CHANNELS_OVERRIDE on mavlink 2.0 does not handle all channels

karthikdesai, 2020-02-13:
### Issue details

Mavlink 2.0 for RC_CHANNELS_OVERRIDE msg (https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE) handles upto 18 channels.
The rc_io.cpp plugin for RC override subscriber accepts only 8 channels. The plugin has to check if mavlink 2.0 is enabled, and then accept other channels.
I could have done a PR to fix this, but i am not sure how to check which mavlink version is currently active (within the plugin context)
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/rc_io.cpp#L229

### MAVROS version and platform

Mavros: HEAD
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ * ] ArduPilot
[  ] PX4

Version: 4.0.0-rc3

vooon, 2020-02-13:
Yes, you just need to update cog.py script and message definition.
You do not have to check for mavlink version, if link is in v1 mode it just won't use extension fields.

karthikdesai, 2020-02-14:
Ah okay thanks for the suggestion.
cog.py is new to me, makes sense to use it for msg generation !

vooon, 2021-09-07:
Fixed by #1616.

closed 2021-09-07 by vooon