mavlink/mavros #1385: Can not connect to mavros

idychenxi, 2020-02-19:
### Issue details

I try to use my computer to connect the mavros in a Raspberry pi but the connection couldn't established.
The ROS_IP and ROS_MASTER_URI for both computer are correct.
I also set /etc/hosts file so each computer can ping each other.
The mavlink version in Raspberry pi is 9.9.2016, so I install this version of mavlink and mavros 0.18.0  on my computer. But there are no connection and no error reported.
Is there any other reason may cause this?

### MAVROS version and platform of my computer

Mavros: 0.18.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[*] ArduPilot
[  ] PX4

### Node logs

```
started roslaunch server http://192.168.1.115:36243/

SUMMARY
========

CLEAR PARAMETERS
 * /follower_1_mavros/

PARAMETERS
 * /follower_1_mavros/cmd/use_comp_id_system_control: False
 * /follower_1_mavros/conn/heartbeat_rate: 1.0
 * /follower_1_mavros/conn/system_time_rate: 0.0
 * /follower_1_mavros/conn/timeout: 10.0
 * /follower_1_mavros/conn/timesync_rate: 0.0
 * /follower_1_mavros/fcu_url: udp://:6002@
 * /follower_1_mavros/gcs_url: 
 * /follower_1_mavros/global_position/frame_id: fcu
 * /follower_1_mavros/global_position/rot_covariance: 99999.0
 * /follower_1_mavros/global_position/tf/child_frame_id: fcu_utm
 * /follower_1_mavros/global_position/tf/frame_id: local_origin
 * /follower_1_mavros/global_position/tf/send: False
 * /follower_1_mavros/image/frame_id: px4flow
 * /follower_1_mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /follower_1_mavros/imu/frame_id: fcu
 * /follower_1_mavros/imu/linear_acceleration_stdev: 0.0003
 * /follower_1_mavros/imu/magnetic_stdev: 0.0
 * /follower_1_mavros/imu/orientation_stdev: 1.0
 * /follower_1_mavros/local_position/frame_id: fcu
 * /follower_1_mavros/local_position/tf/child_frame_id: fcu
 * /follower_1_mavros/local_position/tf/frame_id: local_origin
 * /follower_1_mavros/local_position/tf/send: False
 * /follower_1_mavros/mission/pull_after_gcs: True
 * /follower_1_mavros/mocap/use_pose: True
 * /follower_1_mavros/mocap/use_tf: False
 * /follower_1_mavros/plugin_blacklist: ['actuator_contro...
 * /follower_1_mavros/plugin_whitelist: []
 * /follower_1_mavros/px4flow/frame_id: px4flow
 * /follower_1_mavros/px4flow/ranger_fov: 0.0
 * /follower_1_mavros/px4flow/ranger_max_range: 5.0
 * /follower_1_mavros/px4flow/ranger_min_range: 0.3
 * /follower_1_mavros/safety_area/p1/x: 1.0
 * /follower_1_mavros/safety_area/p1/y: 1.0
 * /follower_1_mavros/safety_area/p1/z: 1.0
 * /follower_1_mavros/safety_area/p2/x: -1.0
 * /follower_1_mavros/safety_area/p2/y: -1.0
 * /follower_1_mavros/safety_area/p2/z: -1.0
 * /follower_1_mavros/setpoint_accel/send_force: False
 * /follower_1_mavros/setpoint_attitude/reverse_throttle: False
[... 257 lines cut ...]
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "follower_1_mavros: Battery"
    message: "No data"
    hardware_id: "udp://:6002@"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"

```

mhallak, 2020-11-11:
Try to remove the modem manager: sudo apt-get remove modemmanager
Then you should be able to connect to your unit.
Note that I would expect a parameter indicating the serial port used but may be you are using default: /dev/ttyACM0 and 57600 baud rate.

vooon, 2020-11-11:
Check your FCU connection URL.

closed 2026-11-11 by github-actions[bot]