mavlink/mavros #1387: Trying to use distance sensor with Typhoon_H480

Pman6543, 2020-02-20:
Hello, 

I am trying to use the distance sensor with the typhoon drone. I want to subscribe to that topic and process the results, but I am having difficulties figuring out how to do it. 

As you can see in the screenshot I have tried playing with the px4_config file, but it results in the error to the right side of the screenshot. 

I have also removed distance sensor and range finder from the blacklist already. 

pxh> [ERROR] [1582220970.124398282, 0.660000000]: DS: gazebo_sonar (id 100) is subscriber, but i got sensor data for that id from FCU

I just want to subscribe to the distance sensor topic in mavros. Thanks!

[image]

TSC21, 2020-02-20:
I am not actually sure why you want to subscribe to the Gazebo range sensor data, considering that that data is already sent from Gazebo to the FCU though the `gazebo_mavlink_interface`.

And the error you get is actually quite explanatory: you are trying to subscribe to a sensor ID 100 when that sensor data already exists in the FCU. Since the FCU publishes the range sensor data on the Mavlink stream as a DISTANCE_SENSOR msg, you are basically trying to republish data that the FCU already has and that is even publishing in Mavlink.

TSC21, 2020-02-20:
One way you can check that data already is being propagated in PX4 is by using the `listener distance_sensor` cmd on the `pxh` console. You will get for sure data from the sonar on the Typhoon model already, so what you are trying to do with MAVROS is no necessary.

Pman6543, 2020-02-20:
I am very new to px4, so I am prone to doing silly things, but I appreciate your response completely. I am trying to program an autonomous application in which the drone detects and moves to an object. 

What is your suggestion for doing this?

TSC21, 2020-02-20:
You opened an issue with respect to a specific question. I advise you to have some readings on the PX4 user and dev guide and push your questions to discuss.px4.io and to the PX4 Slack channels. Closing this as it is answered and solved.

closed 2020-02-20 by TSC21

AlexWUrobot, 2022-12-20:
I find a solution by this : 
https://discuss.px4.io/t/how-to-read-distance-sensor-data-from-mavros-topic/21944