mavlink/mavros #1391: Installation Error; [ament_cmake]

netgittemp, 2020-03-03:
During installation of MAVROS, I run the following command

`
$rosdep install --from-paths src --ignore-src -y
`

I got the following error message.

`
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mavlink: Cannot locate rosdep definition for [ament_cmake]
`

Could you please help me to resolve this issue?

TSC21, 2020-03-03:
What ROS version do you have installed? ament is the build system for ROS2 and we do not currently support MAVROS in ROS2.

vooon, 2020-03-04:
@TSC21 i released mavlink for ROS2, which requires ament (which dep actually may be removed, just i have to do several try-pass releases until got CI working).
But that dependencies have conditions, so it should be fine at least on Melodic.

@netgittemp have you done `source /opt/ros/<ver>/setup.bash` before ran rosdep?

vooon, 2020-03-09:
@netgittemp updates?

jamesthesken, 2020-03-26:
I am encountering the same issue on Ubuntu 18.04 following the instructions from: https://dev.px4.io/v1.10/en/ros/mavros_installation.html

jordancoult, 2020-04-16:
@netgittemp I had the same error for the following setup:

- Ubuntu 16.04
- ROS Kinetic
- mavlink Kinetic 2020.4.4-1

The error went away after I uninstalled and reinstalled ROS Kinetic. However, I probably could've just updated Kinetic instead to get the same result.

vooon, 2020-04-17:
Probably older setup.bash script do not include python version and other vars. Or package reader do not support condition args.

Closing issue as stale. Feel free to reopen.

closed 2020-04-17 by vooon