mavlink/mavros #1393: Cannot set parameter to 0. 'Param types mismatch'

jowaibel, 2020-03-06:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I'm not sure if this is a mavros or px4 issue. That's why I post it on both repos.

It probably appeared for the first time after having updated to PX4 v.1.10.1 stable. I cannot set anymore a parameter to the value 0 via the mavparam set command/service. I tried different parameters, standard and custom ones, their type is float, for example FW_T_RC.
It is however possible to set the param to 0 using QGroundControl.

Running mavparam on the console gives:
```
rosrun mavros mavparam set FW_R_TC 0.0

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/mavros/mavparam", line 104, in <module>
    main()
  File "/opt/ros/melodic/lib/mavros/mavparam", line 100, in main
    args.func(args)
  File "/opt/ros/melodic/lib/mavros/mavparam", line 63, in do_set
    print(param_set(args.param_id, val))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/mavros/param.py", line 149, in param_set
    raise IOError("Request failed.")
IOError: Request failed.
```

Same for this call:

```
rosservice call /mavros/param/set "param_id: 'FW_R_TC'
value:
  integer: 0
  real: 0.0" 
success: False
value: 
  integer: 0
  real: 0.40000000596
```

### MAVROS version and platform

Mavros: ?1.0.0?
ROS: ?Melodic?
Ubuntu: ?18.04.4?

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: ?1.10.1 stable?

### Node logs

```
[ INFO] [1583484719.492793622]: FCU: [pm] param types mismatch param: FW_R_TC
[ WARN] [1583484720.486598977]: PR: Resend param set for FW_R_TC, retries left 2
[ INFO] [1583484720.493704535]: FCU: [pm] param types mismatch param: FW_R_TC
[ WARN] [1583484721.487453377]: PR: Resend param set for FW_R_TC, retries left 1
[ INFO] [1583484721.496680992]: FCU: [pm] param types mismatch param: FW_R_TC
[ WARN] [1583484722.487824199]: PR: Resend param set for FW_R_TC, retries left 0
[ INFO] [1583484722.494284405]: FCU: [pm] param types mismatch param: FW_R_TC
[ERROR] [1583484723.488140772]: PR: Param set for FW_R_TC timed out.
```

### Diagnostics

```
header: 
  seq: 923
  stamp: 
    secs: 1583485318
    nsecs: 570771828
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "/dev/ttyFTDIbridge:921600"
    values: 
      - 
        key: "Received packets:"
        value: "27488"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "194"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "16429038"
      - 
        key: "Tx total bytes:"
        value: "386665"
      - 
        key: "Rx speed:"
        value: "20586.000000"
      - 
        key: "Tx speed:"
        value: "500.000000"
  - 
    level: 2
    name: "mavros: GPS"
    message: "No satellites"
    hardware_id: "/dev/ttyFTDIbridge:921600"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "/dev/ttyFTDIbridge:921600"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "1190"
      - 
        key: "Frequency (Hz)"
        value: "0.999966"
      - 
        key: "Vehicle type"
        value: "Fixed wing aircraft"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "MANUAL"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "/dev/ttyFTDIbridge:921600"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000000"
      - 
        key: "Sensor enabled"
        value: "0x00000000"
      - 
        key: "Sensor health"
        value: "0x00000000"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 2
    name: "mavros: Battery"
    message: "No data"
    hardware_id: "/dev/ttyFTDIbridge:921600"
    values: 
      - 
        key: "Voltage"
        value: "-1.00"
      - 
        key: "Current"
        value: "0.0"
      - 
        key: "Remaining"
        value: "0.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "/dev/ttyFTDIbridge:921600"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "11900"
      - 
        key: "Frequency (Hz)"
        value: "9.999657"
      - 
        key: "Last RTT (ms)"
        value: "8.231912"
      - 
        key: "Mean RTT (ms)"
        value: "6.018272"
      - 
        key: "Last remote time (s)"
        value: "1559.860381000"
      - 
        key: "Estimated time offset (s)"
        value: "1583483758.701039076"
```

### Check ID

```
Received 5314 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 0, 4, 230, 74, 140, 77, 111, 241, 147, 30
```

vooon, 2020-03-06:
That probably both: PX4 require param type to match, but previously casts ints.
And param plugin `integer: 0, real: 0.0` it the special case, which is converted to int=0.

Try to use ros parameters interface + push request.

jowaibel, 2020-03-06:
I agree that this could be the issue. However, it worked previously. So there must be a change in either mavros or px4, which I both updated to the most recent version lately.

Could you please specify what and how exactly to try with the ros params interface?

vooon, 2020-03-06:
Mavros plugins didn't change for a while. So that is because firmware starts to check message more strict.

`rosparam set /mavros/param/FW_R_TC 0.0` (or similar, sorry i do not remember exact syntax).

jowaibel, 2020-03-07:
Yes, thank you, that works. Now, I only need to implement this in a cpp code. I might however change the parameter handling in my code anyway.

In case anyone is wondering, 
`rosparam set /mavros/param/FW_R_TC 0.0` creates a ROS parameter of this name and value.
`rosservice call /mavros/param/push` transfers all ROS parameters of the name /mavros/param/... to the px4 parameter database (if the corresponding name exists).

closed 2026-11-10 by github-actions[bot]