mavlink/mavros #1396: Different topics rate on same computers

elgarbe, 2020-03-08:
Hi, I have to vehicles with pixhawk, serial connection to a raspberry pi 3b+, booth FC with the same configuration in telem2 port (Mavlink and 921600bps). I have a wifi router and booth vehicles and my laptop on the same network and in the same place. I run apm.laucnh on each raspi and rosrun mavsys rate 30. In each raspberry I run rostopic hz /mavros/global_position/local and in one of them I get 22 Hz and on the other 17 Hz. What could be the source for different topic rates?

Thank!

amilcarlucas, 2020-10-09:
Is the publish rate of the Mavlink messages the same on both systems?

elgarbe, 2020-10-09:
> Is the publish rate of the Mavlink messages the same on both systems?

As I sed I set the mavros rate to 30Hz on both Raspberry computer. I don't know if I can set publish rate on pixhawk, I don't think so.

Thank

vooon, 2020-10-09:
Maybe it's depends on GPS module. Can you try to check some other message, that is not dependant on external sensors.
What about IMU messages? Also please compare CPU/RAM usages on both rpi's.

elgarbe, 2020-10-09:
I will check it tomorrow and let you know the results. BTW both GPS are ublox M8N module. The hardware is the same on low/high level computer/controller.

closed 2026-11-10 by github-actions[bot]