mavlink/mavros #1397: [ERROR] [1583738666.486151358, 65.500000000]: 1: DROPPED Message-Id 42 [1 bytes] IDs: 1.1 Seq: 80: MAVConnSerial::send_message: TX queue overflo

maerad7, 2020-03-09:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_
[image]
============================================================================
The companion computer was setup as above.
And GCS was connected through MAVROS.

   fcu_url :="udp://:14540@localhost:14557"
   gcs_url:="serial:///dev/ttyUSB0:57600" 
There is some problem with the connection as this error message is being displayed .

log:companion computer
ERROR [mavlink] instance 1: RADIO_STATUS timeout

Following popup messages is displayed on QGC.

Vehicle 1 did not respond to request for parameters. This will cause QGC to be unable to display its full user interface.

when mission is uploaded, following message is displayed:

Mission transfer faileld.Retry transfer.
Error : Mission write failed, vehicle failed to send final ack.
And ideas with how this can be solved?

### MAVROS version and platform

Mavros: git master viersion compile version 1.0?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

PX4

Version: 10.0.0

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
==========================================================================
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
process[mavros-5]: started with pid [24448]
INFO  [param] selected parameter default file eeprom/parameters_1015
[param] Loaded: eeprom/parameters_1015
process[gazebo-6]: started with pid [24458]
process[spawn_model-7]: started with pid [24472]
process[local_planner_manager-8]: started with pid [24490]
process[local_planner_nodelet-9]: started with pid [24499]
process[rviz-10]: started with pid [24504]
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
[ INFO] [1583738561.972824186]: Loading nodelet /local_planner_nodelet of type LocalPlannerNodelet to manager local_planner_manager with the following remappings:
[ INFO] [1583738561.979220608]: waitForService: Service [/local_planner_manager/load_nodelet] has not been advertised, waiting...
INFO  [commander] Mission #2 loaded, 4 WPs, curr: 3
[ INFO] [1583738561.989313969]: Initializing nodelet with 12 worker threads.
[ INFO] [1583738562.023057917]: FCU URL: udp://:14540@localhost:14557
[ INFO] [1583738562.024830591]: udp0: Bind address: 0.0.0.0:14540
[ INFO] [1583738562.024904061]: udp0: Remote address: 127.0.0.1:14557
[ INFO] [1583738562.025064676]: GCS URL: serial:///dev/ttyUSB0:57600
[ INFO] [1583738562.025120639]: serial1: device: /dev/ttyUSB0 @ 57600 bps
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
[... 322 lines cut ...]
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "8.068000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.832000045"

```

### Check ID

```
rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 6735 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 30, 31, 32, 33, 36, 42, 62, 76, 74, 331, 332, 83, 340, 85, 87, 230, 105, 111, 241, 242, 245

```

vooon, 2020-03-09:
As i see, you setup serial conn for gcs, it uses conn id 1, so DROPPED error related to that port.
And it may be simply because message rate from PX4 simulator much higher than possible to send via radio.

```
        key: "Rx speed:"
        value: "inf"
```
Strange... But it already ~5 MiB, which would take a while to transmit on the speed of 5760 Byte/sec.

maerad7, 2020-03-10:
thank you for telling me. I will try in the drone.

dongsin-kim, 2020-06-15:
Hi @maerad7 ,

Did you manage this issue? I also think that this is related to the message rate.

I also experienced the same issue and fixed it by controlling the _while()_ using a **`ros::spin()`**.

closed 2026-11-10 by github-actions[bot]