mavlink/mavros #1398: setpoint_raw/local on slow velocity

bavokn, 2020-03-10:
### Issue details

Hi, i'm having some difficulties using the topic setpoint_raw/local. 

I'm using gazebo as simulation. I first set the drone in offboard mode before flying 

I'm trying to let a drone go from point to point but at a slow velocity. I'm trying do do this by creating a TargetPosition object and setting the velocity x,y to 0.2 like this :

```
def get_PositionTarget(self, x , y , z ):
        target = PositionTarget()
        target.coordinate_frame = PositionTarget.FRAME_LOCAL_NED
        #setting position
        target.position.x = x
        target.position.y = y
        target.position.z = z
        #setting vel
        target.velocity.x = 0.2
        target.velocity.y = 0.2
        target.velocity.z = 0
        target.type_mask = target.IGNORE_AFX | target.IGNORE_AFY | target.IGNORE_AFZ 

        return target
``` 
I do some calculations to calculate points on the map using local coordinates and then save them using this function in self.points

later i set the header.stamp before actually publishing

```
self.points[current].header.stamp = rospy.Time.now()
self.point_push.publish(self.points[current])
```

The drone flies from point to point perfectly but way to fast, how can i set the velocity so that the drone goes at a very slow pace between the points ?

i've tried using diffrent masks and trying with acceleration_or_force but whatever i set as values nothing changes the speed of the drone between 2 points.

### MAVROS version and platform

Mavros: 1.0.0
ROS: melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: ?3.7.1?

### /iris_0/mavros/setpoint_raw/local log
here you can see the drone fly from point to point and it receives the velocity correctly but in the simulation it still goes at full speed
```
header: 
  seq: 85
  stamp: 
    secs: 1395
    nsecs: 413000000
  frame_id: ''
coordinate_frame: 1
type_mask: 3552
position: 
  x: 10.5434504467
  y: 1.50620720668
  z: 5.0
velocity: 
  x: 0.2
  y: 0.2
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 0.0
yaw_rate: 0.0
---
header: 
  seq: 86
  stamp: 
    secs: 1395
    nsecs: 746000000
  frame_id: ''
coordinate_frame: 1
type_mask: 3552
position: 
  x: 13.5558648601
  y: 1.50620720668
  z: 5.0
velocity: 
  x: 0.2
  y: 0.2
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 0.0
yaw_rate: 0.0
---
header: 
  seq: 87
  stamp: 
    secs: 1396
    nsecs:  82000000
  frame_id: ''
coordinate_frame: 1
type_mask: 3552
position: 
  x: 13.4937927933
  y: 28.4937927933
  z: 5.0
velocity: 
  x: 0.2
  y: 0.2
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 0.0
yaw_rate: 0.0
---
header: 
  seq: 88
  stamp: 
    secs: 1396
    nsecs: 414000000
  frame_id: ''
coordinate_frame: 1
type_mask: 3552
position: 
  x: 13.4937927933
  y: 28.4937927933
  z: 5.0
velocity: 
  x: 0.2
  y: 0.2
  z: 0.0
acceleration_or_force: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 0.0
yaw_rate: 0.0

```

### Diagnostics mid flight 

```
header: 
  seq: 107
  stamp: 
    secs: 114
    nsecs:   3000000
  frame_id: ''
status: 
  - 
    level: 0
    name: "iris_1/mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Received packets:"
        value: "47094"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "246"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "2764885"
      - 
        key: "Tx total bytes:"
        value: "64386"
      - 
        key: "Rx speed:"
        value: "24430.000000"
      - 
        key: "Tx speed:"
        value: "766.000000"
  - 
    level: 0
    name: "iris_1/mavros: GPS"
    message: "3D fix"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Satellites visible"
        value: "10"
      - 
        key: "Fix type"
        value: "3"
      - 
        key: "EPH (m)"
        value: "0.00"
      - 
        key: "EPV (m)"
        value: "0.00"
  - 
    level: 0
    name: "iris_1/mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "113"
      - 
        key: "Frequency (Hz)"
        value: "0.999818"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "OFFBOARD"
      - 
        key: "System status"
        value: "Active"
  - 
    level: 2
    name: "iris_1/mavros: System"
    message: "Sensor health"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Sensor present"
        value: "0x0020002F"
      - 
        key: "Sensor enabled"
        value: "0x0021000F"
      - 
        key: "Sensor health"
        value: "0x0020002F"
      - 
        key: "3D gyro"
        value: "Ok"
      - 
        key: "3D accelerometer"
        value: "Ok"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "rc receiver"
        value: "Fail"
      - 
        key: "AHRS subsystem health"
        value: "Ok"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "iris_1/mavros: Battery"
    message: "Normal"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Voltage"
        value: "12.02"
      - 
        key: "Current"
        value: "-1.0"
      - 
        key: "Remaining"
        value: "93.0"
  - 
    level: 0
    name: "iris_1/mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "1132"
      - 
        key: "Frequency (Hz)"
        value: "9.997273"

```

TSC21, 2020-03-10:
@bavokn you cannot do both position and velocity offboard control to the same axis - you do one or you do the other, but not both to the same axis. If you want to do position control but limit the horizontal velocity, then adjust the [`MPC_XY_VEL_MAX`](https://dev.px4.io/master/en/advanced/parameter_reference.html#MPC_XY_VEL_MAX) and [`MPC_XY_CRUISE`](https://dev.px4.io/master/en/advanced/parameter_reference.html#MPC_XY_CRUISE) parameters.

closed 2020-03-10 by TSC21

Rezenders, 2020-05-14:
How can this be set with launch files?
I tried:
```
  <include file="$(find px4)/launch/mavros_posix_sitl.launch">
      <env name="MPC_XY_CRUISE" value="3.1"/>
      <env name="MPC_XY_VEL_MAX" value="20"/>
      <arg name="respawn_gazebo" value="true"/>
      <arg name="respawn_mavros" value="true"/>
  </include>
```
But it does not work