mavlink/mavros #1400: distance_sensor publisher. Incorrect orientation on the ROS side

standmit, 2020-03-19:
### Issue details

The distance_sensor plugin reads orientation of rangefinder from mavlink message and this orientation is used for sending TF. But it is assumed that orientation in mavlink message is valid for NED frame not front-left-up. In case of ArduCopter, if parameter RNGFND1_ORIENT set to 25 (Down), in rviz rangefinder will be directed up. 

### MAVROS version and platform

Mavros: 1.0.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: 4.0.0

### Plugin confiugration
```yaml
distance_sensor:
    rf_down:
        subscriber: false
        id: 0
        field_of_view: 0.007
        frame_id: "rfd_down"
        send_tf: true
        sensor_position:
            x: 0.0
            y: 0.11
            z: -0.06
```
[image]

dayjaby, 2020-05-02:
Can you add orientation: PITCH_270 to the yaml?

```
distance_sensor:
    rf_down:
        subscriber: false
        id: 0
        field_of_view: 0.007
        frame_id: "rfd_down"
        orientation: PITCH_270
        send_tf: true
        sensor_position:
            x: 0.0
            y: 0.11
            z: -0.06
```
Both sides (ArduPilot and mavros) need to know the orientation of the sensor. You can find the names for the orientations in https://mavlink.io/en/messages/common.html#MAV_SENSOR_ROTATION_PITCH_270

standmit, 2020-05-03:
@dayjaby I mean another problem. `PITCH_270` mean "**down**", but in fact frame points **up**. If I will try to override it by value `PITCH_90`, the plugin will emit error message `received sensor data has different orientation (PITCH_270) than in config (PITCH_90)!`

standmit, 2020-05-04:
The problem can be reproduced using package: https://github.com/standmit/dist_err

mortenfyhn, 2021-05-03:
I have the same issue. An example to illustrate further:

* My autopilot sends a MAVLink DISTANCE_SENSOR with orientation 25 (MAV_SENSOR_ROTATION_PITCH_270)
* MAVROS must be configured for PITCH_270, otherwise it emits a "wrong orientation" error
* If I enable `send_tf`, then a ROS tf matching PITCH_270 will be created

However, 270 deg pitch is only correct in autopilot frame (forward-right-down), not in ROS frame (forward-left-up). In ROS frame, pitch 90 would be correct. Therefore, my downward facing sensor shows up as upward facing in Rviz.

As far as I can tell this is still the case on mavros 1.7.1.

closed 2026-11-10 by github-actions[bot]