mavlink/mavros #1404: trying to control PX4 by setting /mavros/setpoint_velocity/cmd_vel_unstamped with mavros from Jetson Xavier, msg publication works fine but robot does not move.

YB17, 2020-03-25:
### Issue details

Background: I'm trying to control a holybro PX4 by Jetson Xavier with mavros for my rover indoors (therefore with no GPS signal) by setting the /mavros/setpoint_velocity/cmd_vel_unstamped message. The PX4 board is connected with the Jetson Xavier with a USB cable. For the px4 I used: 

roslaunch mavros px4.launch 

with properly setting the tty device name, and the ROS info is below. px4_config.yaml is used without any modification, and therefore "mav_frame" in px4_config.yaml is set to LOCAL_NED.

Now the RC works fine in the sense that in manual mode I can control the rover with no problem, and the demo code "offb_node.cpp" from the px4.io works fine in the sense that it can set the PX4 to armed and offboard mode correctly as specified in the source code. 

Problem: when I am trying to modify the code in offb_node.cpp to publish velocity data to /mavros/setpoint_velocity/cmd_vel_unstamped, the message publication works fine (as I can see the velocity linear.x, angular.z changing with rostopic echo), however the rover has no reaction at all. Any idea on the issue is welcome, Thanks!!!

### MAVROS version and platform

Mavros: 1.0.0
ROS: melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.10 - stable

### Node logs

```
... logging to /home/yb/.ros/log/fe2ab722-6ea7-11ea-9b07-9e8577235b45/roslaunch-yb-desktop-31732.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yb-desktop:34231/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
[... 427 lines cut ...]
        key: "Last RTT (ms)"
        value: "0.871012"
      - 
        key: "Mean RTT (ms)"
        value: "0.598333"
      - 
        key: "Last remote time (s)"
        value: "512.221532000"
      - 
        key: "Estimated time offset (s)"
        value: "1585147823.888042688"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 6392 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 26, 30, 31, 32, 36, 42, 65, 69, 74, 331, 77, 340, 230, 105, 111, 241, 116, 245
```

YB17, 2020-04-05:
Never mind. I solved the problem myself days ago.

closed 2020-04-05 by YB17

eleclong, 2020-06-06:
hello？
Is there any new progress？
I encountered the same problem as you

YB17, 2020-07-05:
Yep in case you are wondering, the navigation stack works fine and I am working on connecting some short-range sensors via I2C A of PX4, which is not relevant to the question.

For the problem, there could be many potential causes. You need to debug the code carefully to see if everything works fine on the PX4 end as well as the actuators. In my case I found that the PX4 offboard won't work if there is no valid GPS signal. Since my robot is indoor, there is no GPS used. Hope it helps although it is a bit late.