mavlink/mavros #1411: setpoint_position documentation outdated

mstniy, 2020-04-11:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

It seems with #1306 `~setpoint_position/global` switched from accepting a `mavros_msgs/GlobalPositionTarget` to a `geographic_msgs::GeoPoseStamped`.

However, the documentation at [https://wiki.ros.org/mavros](https://wiki.ros.org/mavros) does not reflect this change.

### MAVROS version and platform

Mavros: 1.0.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.9.2

### Node logs

```
[ERROR] [1586623417.227640692]: Client [/mavros] wants topic /mavros/setpoint_position/global to have datatype/md5sum [geographic_msgs/GeoPoseStamped/cc409c8ed6064d8a846fa207bf3fba6b], but our version has [mavros_msgs/GlobalPositionTarget/076ded0190b9e045f9c55264659ef102]. Dropping connection.
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2020-04-11:
Please free to update the wiki in that direction. The wiki docs are outdate since a long time and I personally don't have time to update them.

closed 2020-04-11 by TSC21

lyuuuuuu, 2020-05-27:
I got an error like :client wants topic /mavros/setpoint_position/global to have datatype/md5sum [geographic_msgs/geo pose stamped]...but our version has [mavros_msgs/global position target/...].dropping connection.  
Is that the reason?
( I use `mavros_msgs::GlobalPositionTarget global_position_msgs` in my node.

Coderx7, 2021-08-08:
@lyuuuuuu did you solve this?

murphym18, 2021-09-08:
I updated the ROS wiki to bring the setpoint_position section up to date. Should be solved now :+1: