mavlink/mavros #1412: mavros commands about controlling orientation , acceleration and angular speed didn't work

unravelwool, 2020-04-20:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I'm using Mavros to communicate with Ardupilot SITL to control my UAV in Gazebo simulator. When I test the mavros methods of controlling uav by publishing some messages to `mavros/setpoint_xxx/xxxx` topic, I find some controlling comands do not work. Please see the following description for details.
  1) Publish messages to `mavros/setpoint_accel/accel` topic, the uavs in both ardupilot SITL and Gazebo  don't move, the command controlling acceleration doesn't work.
  2) Publish messgaes to `mavros/setpoint_attitude/cmd_vel` topic, the uavs in both ardupilot SITL and Gazebo don't move.
  3) Publish messgaes to `mavros/setpoint_position/local` topic, the command set the target position and orientation the uav get. However, the uav in simulation can only translate to the target position, but does not achieve  the target orientation required by the command.
  4) Publish messages to `mavros/setpoint_velocity/cmd_vel` topic, the messgaes set the value of x/y/z axis linear speed and roll/pitch/yaw angular speed, but the uav in simulation can only fly in translation method, and don't rotate.
above mavros command  sent to apm sitl  is all in `guided mode` (a flight mode in apm) and after takeoff.
Summary: the mavros can set the position and x/y/z axis speed of uav, but can not control the angles and angular speeds of uav.

Is something about my mavros wrong? or some param about mavros is set wrong? How can I  control the yaw angle and angular speed?
Besides, I am let mavros communicate to Ardupilot SITL using the `apm_sitl.launch` file like this:
```
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- example launch script for ArduPilotMega based FCU's  -->

	<arg name="fcu_url" default="udp://127.0.0.1:14550@" />
	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />

	<include file="$(find mavros)/launch/node.launch">
		<arg name="pluginlists_yaml" value="$(find mavros)/launch/apm_pluginlists.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/apm_config.yaml" />
		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
		<arg name="log_output" value="$(arg log_output)" />
	</include>
</launch>
```

### MAVROS version and platform

Mavros: ?0.18.4? maybe, I have forget the version of mavros. how I can get the version information.
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: Copter V3.6-dev (c6a0763b)

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /group1/mavros/

PARAMETERS
 * /group1/mavros/cmd/use_comp_id_system_control: False
 * /group1/mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /group1/mavros/conn/heartbeat_rate: 1.0
 * /group1/mavros/conn/system_time_rate: 1.0
 * /group1/mavros/conn/timeout: 10.0
 * /group1/mavros/conn/timesync_rate: 0.0
 * /group1/mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
[... 395 lines cut ...]
    hardware_id: "udp://127.0.0.1:14550@"
    values: 
      - 
        key: "Voltage"
        value: "12.19"
      - 
        key: "Current"
        value: "28.7"
      - 
        key: "Remaining"
        value: "61.0"
---
```

### Check ID

```
rosrun mavros checkid
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

PotatoesLegend, 2020-06-28:
Hi @unravelwool ,
Did you progress on this task?
I'm also trying to control a UAV from sitl with mavros and gazebo simulation.

mariofaragalla, 2021-02-18:
I had similar issues and here is my conclusion:

1- For the topic **/mavros/setpoint_position/global** :I checked the type of messages published on it by running `rostopic info /mavros/setpoint_position/global ` I found out that the message type that I should be working with is **geographic_msgs/GeoPoseStamped** not **mavros_msgs/GlobalPositionTarget** (I don't know why the [documentation](https://wiki.ros.org/mavros#mavros.2FPlugins.setpoint_position) is not updated) and this worked for me.

2- For the topic  **mavros/setpoint_attitude/cmd_vel** : it didn't work for the velocity but the topic `mavros/setpoint_velocity/cmd_vel` (which is not mentioned in the documentation but can be found in the setpoint.py file which can be found in /catkin_ws/src/mavros/mavros/src/mavros ) worked for me.

3- For the topic **mavros/setpoint_accel/accel** : unfortunately I am still trying to figure out how to make it work

closed 2026-11-09 by github-actions[bot]