mavlink/mavros #1417: frame_id for setpoint commands?

sayali-purdue, 2020-04-28:
I am statically analyzing the ROS based code. While doing so, I observed that many ROS nodes that publish setpoints to MavRos Topics do not assign any frame to the frame_id field of the message. Therefore, I would really appreciate if you could provide the following details:

1. What frame should be set in the frame_id field of the setpoint command message, e.g. poseStamped message published on the mavros/setpoint_position/local topic? 

2. Does MavRos ensure that the incoming message is in the correct/expected frame? If it does not, then why doesn't it enforce the correct frame utilization by the developers of the nodes that use MavRos?

Thank you!

TSC21, 2020-04-28:
1. @sayali-purdue check https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_position.cpp#L139-L154 and https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_position.cpp#L281-L284. The `frame_id` is irrelevant in this case because it depends only on the `mav_frame` variable (or if you are listening to a tf, look at https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_position.cpp#L56-L57).
2. We do not enforce frames of reference anywhere since when a TF is expected between two frames of reference, an error will pop signalling a problem in the TF tree. Otherwise, the frame_id is irrelevant, since you are using the data to parse to Mavlink messages.

sayali-purdue, 2020-04-28:
Thanks a lot for your reply!

1. So, as per the code, MavRos just assumes that the input setpoint is specified in any frame that follows ENU (or FLU for body-frame) convention? For example, in case of body frame, it can be either base_link, or base_footprint, or other with FLU convention?

2. 
> We do not enforce frames of reference anywhere since when a TF is expected between two frames of reference, an error will pop signalling a problem in the TF tree.

Can you please elaborate on this (especially TF between which two frames of reference)? Do you mean the frame of the input setpoint message and the expected frame of MavRos?

Thanks again!

TSC21, 2020-04-28:
1. As I said, it's irrelevant for the code per-si. But yes it is the responsibility of the developer to make sure that any data sent in the setpoints is in the correct frame of reference.

2. I sent an example above: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_position.cpp#L56-L57. I mean the TF tree existing frames, doesn't have to be the `frame_id` in the messages.

sayali-purdue, 2020-04-28:
2. As I understand from the example https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_position.cpp#L56-L57, the TF between target position and map frames published by the developer is an alternative way to send setpoint command. This TF basically provides the target position in the map reference frame. Is that right?

But I still do not understand the following:

> We do not enforce frames of reference anywhere since when a TF is expected between two frames of reference, an error will pop signaling a problem in the TF tree

What error might occur in the TF tree by enforcing frame of reference anywhere else?

TSC21, 2020-04-28:
> What error might occur in the TF tree by enforcing frame of reference anywhere else?

I am not saying an error occurs if you enforce the frame anywhere else. An error will pop if one of the frames doesn't exist or if there is not connection between the parent and child frames that one is setting. This is common to all usages of tf and tf lookups in ROS, not something specific to MAVROS.

closed 2026-11-09 by github-actions[bot]