mavlink/mavros #1418: How to install mavros for ros melodic & ubuntu 18.04

staycoolish, 2020-04-30:
Hello,

I cannot install mavros, in Ubuntu 18.04, using the command

**sudo apt-get install ros-melodic-mavros ros-melodic-mavros-extras**

It says

**E: Unable to locate package ros-melodic-mavros
E: Unable to locate package ros-melodic-mavros-extras**

What should I do? 

Thanks,

TSC21, 2020-04-30:
@staycoolish how did you install ROS Melodic?

Jaeyoung-Lim, 2020-04-30:
@staycoolish Did you install ROS?

staycoolish, 2020-04-30:
Actually I followed the steps from http://wiki.ros.org/melodic/Installation/Ubuntu , but I might have removed something by accident, every time I start Ubuntu, a system error (with Report it / Cancel options) is given to me. Do you think uninstalling & reinstalling ROS might work?

Jaeyoung-Lim, 2020-04-30:
@staycoolish It is not about reinstalling - you are missing the ros repository from the sources list. If you were able to install ROS, you should be able to install mavros

staycoolish, 2020-04-30:
Thank you, @TSC21 and @Jaeyoung-Lim ! I have applied the installation processes again (which took a very short time), from http://wiki.ros.org/melodic/Installation/Ubuntu , then I was able to apply apt-get install commands, for mavros and some other packages.

closed 2020-04-30 by staycoolish