mavlink/mavros #1422: FCU: DeviceError:serial:open

Bertrand731, 2020-05-06:
Hello,

I have the following issue when trying to launch Mavros

### Issue details

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

[FATAL] [1588758984.637796557]: FCU: DeviceError:serial:open: No such file or directory
================================================================================REQUIRED process [mavros-2] has died!
process has finished cleanly
```
place here result of:
rostopic echo -n1 /diagnostics
```
ERROR: Unable to communicate with master!

### Check ID

```
rosrun mavros checkid
```
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/mavros/checkid", line 109, in <module>
    main()
  File "/opt/ros/kinetic/lib/mavros/checkid", line 103, in main
    rospy.init_node("checkid", anonymous=True)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 336, in init_node
    raise rospy.exceptions.ROSInitException("Failed to initialize time. Please check logs for additional details")
rospy.exceptions.ROSInitException: Failed to initialize time. Please check logs for additional details

vooon, 2020-05-07:
What fcu_url is set? Check `ls /dev/tty*`.

Bertrand731, 2020-05-07:
I have the following list :
/dev/tty    /dev/tty23  /dev/tty39  /dev/tty54      /dev/ttyS10  /dev/ttyS26
/dev/tty0   /dev/tty24  /dev/tty4   /dev/tty55      /dev/ttyS11  /dev/ttyS27
/dev/tty1   /dev/tty25  /dev/tty40  /dev/tty56      /dev/ttyS12  /dev/ttyS28
/dev/tty10  /dev/tty26  /dev/tty41  /dev/tty57      /dev/ttyS13  /dev/ttyS29
/dev/tty11  /dev/tty27  /dev/tty42  /dev/tty58      /dev/ttyS14  /dev/ttyS3
/dev/tty12  /dev/tty28  /dev/tty43  /dev/tty59      /dev/ttyS15  /dev/ttyS30
/dev/tty13  /dev/tty29  /dev/tty44  /dev/tty6       /dev/ttyS16  /dev/ttyS31
/dev/tty14  /dev/tty3   /dev/tty45  /dev/tty60      /dev/ttyS17  /dev/ttyS4
/dev/tty15  /dev/tty30  /dev/tty46  /dev/tty61      /dev/ttyS18  /dev/ttyS5
/dev/tty16  /dev/tty31  /dev/tty47  /dev/tty62      /dev/ttyS19  /dev/ttyS6
/dev/tty17  /dev/tty32  /dev/tty48  /dev/tty63      /dev/ttyS2   /dev/ttyS7
/dev/tty18  /dev/tty33  /dev/tty49  /dev/tty7       /dev/ttyS20  /dev/ttyS8
/dev/tty19  /dev/tty34  /dev/tty5   /dev/tty8       /dev/ttyS21  /dev/ttyS9
/dev/tty2   /dev/tty35  /dev/tty50  /dev/tty9       /dev/ttyS22
/dev/tty20  /dev/tty36  /dev/tty51  /dev/ttyprintk  /dev/ttyS23
/dev/tty21  /dev/tty37  /dev/tty52  /dev/ttyS0      /dev/ttyS24
/dev/tty22  /dev/tty38  /dev/tty53  /dev/ttyS1      /dev/ttyS25

Bertrand731, 2020-05-11:
> What fcu_url is set? Check `ls /dev/tty*`.
I have the following list :
/dev/tty /dev/tty23 /dev/tty39 /dev/tty54 /dev/ttyS10 /dev/ttyS26
/dev/tty0 /dev/tty24 /dev/tty4 /dev/tty55 /dev/ttyS11 /dev/ttyS27
/dev/tty1 /dev/tty25 /dev/tty40 /dev/tty56 /dev/ttyS12 /dev/ttyS28
/dev/tty10 /dev/tty26 /dev/tty41 /dev/tty57 /dev/ttyS13 /dev/ttyS29
/dev/tty11 /dev/tty27 /dev/tty42 /dev/tty58 /dev/ttyS14 /dev/ttyS3
/dev/tty12 /dev/tty28 /dev/tty43 /dev/tty59 /dev/ttyS15 /dev/ttyS30
/dev/tty13 /dev/tty29 /dev/tty44 /dev/tty6 /dev/ttyS16 /dev/ttyS31
/dev/tty14 /dev/tty3 /dev/tty45 /dev/tty60 /dev/ttyS17 /dev/ttyS4
/dev/tty15 /dev/tty30 /dev/tty46 /dev/tty61 /dev/ttyS18 /dev/ttyS5
/dev/tty16 /dev/tty31 /dev/tty47 /dev/tty62 /dev/ttyS19 /dev/ttyS6
/dev/tty17 /dev/tty32 /dev/tty48 /dev/tty63 /dev/ttyS2 /dev/ttyS7
/dev/tty18 /dev/tty33 /dev/tty49 /dev/tty7 /dev/ttyS20 /dev/ttyS8
/dev/tty19 /dev/tty34 /dev/tty5 /dev/tty8 /dev/ttyS21 /dev/ttyS9
/dev/tty2 /dev/tty35 /dev/tty50 /dev/tty9 /dev/ttyS22
/dev/tty20 /dev/tty36 /dev/tty51 /dev/ttyprintk /dev/ttyS23
/dev/tty21 /dev/tty37 /dev/tty52 /dev/ttyS0 /dev/ttyS24
/dev/tty22 /dev/tty38 /dev/tty53 /dev/ttyS1 /dev/ttyS25

vooon, 2020-05-12:
Ok, but what is your fcu url?
Default one is `/dev/ttyACM0:57600` (arises when FCU connected directly trough USB), and i do not see it in the list.

KimSeongheon678, 2021-06-10:
> Ok, but what is your fcu url?
> Default one is `/dev/ttyACM0:57600` (arises when FCU connected directly trough USB), and i do not see it in the list.

i have the same problem . 
so what should i do next??

vooon, 2021-06-10:
You need to provide `fcu_url` argument to the real serial device that you have.

https://github.com/mavlink/mavros/blob/5f706efd4102fa2ec0b0aa858c281e528441c67a/mavros/launch/px4.launch#L5

Closing ticket as stall.

closed 2021-06-10 by vooon