mavlink/mavros #1423: Mavros connects to FCU but it won't take any command - timeout

rsisto, 2020-05-07:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi, I'm having issues connecting using mavros to ardupilot rover.
Using APM Planner the connection works great, but when trying over mavros I can receive data from the device, but timeout occurs when I want to send (for example, I can't change the mode).

Example, issuing this command replies ok:
rosservice call /mavros/set_mode 0 AUTO
mode_sent: True

After checking /diagnostics topic, the mode is still:
        key: "Mode"
        value: "HOLD"

And I keep seeing messages of this sort in mavros:
[ INFO] [1588833010.541105240]: HP: requesting home position
[ WARN] [1588833015.545757831]: CMD: Command 410 -- wait ack timeout

Any help will be appreciated, thanks!!

### MAVROS version and platform

Mavros: 1.0.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[ X] ArduPilot
[  ] PX4

### Node logs

```
ubuntu@ubuntu-VPCCW21FX:~/workspaces/hidrodata/hidrodata_heron$ roslaunch mavros apm.launch fcu_url:=/dev/ttyUSB0:57600 fcu_protocol:="v2.0"
... logging to /home/ubuntu/.ros/log/c2d08ec4-9029-11ea-9e88-2c8158f2e67b/roslaunch-ubuntu-VPCCW21FX-12995.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-VPCCW21FX:32853/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
[... 442 lines cut ...]
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
```

### Check ID

```
rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 668 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 163, 36, 165, 33, 42, 24, 35, 62, 150, 152, 27, 74, 30

```

vooon, 2020-05-07:
You have a problem somewhere on computer Tx -- FCU Rx connection.

rsisto, 2020-05-08:
Hi @vooon , thanks for your reply. 
I realised this only happens with mavros, so I think I'm missing some configuration.. 
Using APM Planner (on the same Ubuntu host) works flawlessly, no errors and connects instantly.

Can you suggest some parameters to check to keep looking into this?

Thanks, I appreciate your comments!!

vooon, 2020-05-08:
Have you had apm planner running, while trying to start mavros?
From logs i see that message flow from autopilot to mavros works, but nothing gets to the autopilot from mavros.
Usually that is hardware problem, e.g. bad connection, not connected grounds etc.
But serial port may be blocked by other software. So need to check if you can connect with something else, e.g. mavproxy to /dev/ttyUSB0 with 57600 baud rate.
Also maybe you configured fcu's port for some other function than mavlink (v2).

rsisto, 2020-05-08:
Hi, thanks for your quick answer!!

Regarding your questions, I'm confident the port is free for mavros.
From a fresh startup, I run:
**first terminal**
```
$ roscore
... logging to /home/ubuntu/.ros/log/23bf80f8-916c-11ea-811d-2c8158f2e67b/roslaunch-ubuntu-VPCCW21FX-14585.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-VPCCW21FX:33405/
ros_comm version 1.12.14

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES

auto-starting new master
process[master]: started with pid [14595]
ROS_MASTER_URI=http://ubuntu-VPCCW21FX:11311/

setting /run_id to 23bf80f8-916c-11ea-811d-2c8158f2e67b
process[rosout-1]: started with pid [14612]
started core service [/rosout]

```
**second terminal**

```
~$ roslaunch mavros apm.launch fcu_url:=/dev/ttyUSB0:57600 
... logging to /home/ubuntu/.ros/log/23bf80f8-916c-11ea-811d-2c8158f2e67b/roslaunch-ubuntu-VPCCW21FX-14596.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu-VPCCW21FX:36019/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
[... 469 lines cut ...]
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
 MAV> GPS lock at 51 meters
online system 1
ARMED
MANUAL> Mode MANUAL
Arming checks disabled
fence breach
waypoint 1

MANUAL> 
MANUAL> wp
MANUAL> usage: wp <editor|list|load|update|save|set|clear|loop|remove|move|movemulti|changealt>
wp list
MANUAL> log list
MANUAL> Requesting log list
auto
MANUAL> 

```

This seems a good question for mavlink team as well then :)

Any other thoughts? Everything works great with APM Planner... I assume it uses the same software stack beneath..
Can you suggest please some other tool to use, or command, config?

rsisto, 2020-05-20:
Hi, just a quick follow up...
I switched to a second antenna and same thing happening.

Looking up I found a mavproxy argument that helped:
https://discuss.ardupilot.org/t/raspberry-pi-not-sending-information-to-pixhawk/12272/8

`mavproxy.py --mav10`

This command now enables me to switch mode without any issues.
I'll continue the path back to mavros and will upload my results.

rsisto, 2020-06-01:
@vooon do you have any insight for this? I need to move forward, my plan now is to start building a mavproxy.py wrapper, but that's what mavros does, right?
It's definitely some tweaking in mavros that I'm missing.

Thanks!

vooon, 2020-06-03:
Since MAVLink 1.0 works, you may try to set `~fcu_protocol` = `v1.0`.
See: https://github.com/mavlink/mavros/blob/master/mavros/src/lib/mavros.cpp#L55

But i recommend to check firmware parameters and change to MAVLink 2.0.

Also to eliminate possible wiring problem, try to connect to FCU's USB and use `/dev/ttyACM0`.

rsisto, 2020-06-27:
Hi, thanks as always for your answer!

I checked and there's no wiring problems, since mavproxy works great.
I even ran a mavproxy as daemon as suggested in https://ardupilot.org/dev/docs/making-a-mavlink-wifi-bridge-using-the-raspberry-pi.html
Connecting a second mavproxy via udp works great as well.

But when trying mavros (even in this scenario) I step with timeout issues and no message being sent.

[ INFO] [1593268904.634011131]: udp0: Remote address: 127.0.0.1:14550
[ INFO] [1593268904.826430207]: IMU: Raw IMU message used.
[ WARN] [1593268904.826560056]: IMU: linear acceleration on RAW_IMU known on APM only.
[ WARN] [1593268904.826610975]: IMU: ~imu/data_raw stores unscaled raw acceleration report.
[ INFO] [1593268905.552538558]: CON: Got HEARTBEAT, connected. FCU: ArduPilot
[ WARN] [1593268907.557015277]: VER: broadcast request timeout, retries left 4
[ WARN] [1593268908.557748588]: VER: broadcast request timeout, retries left 3
[ WARN] [1593268914.557073756]: CMD: Command 520 -- wait ack timeout
[ WARN] [1593268914.557387545]: VER: unicast request timeout, retries left 2
[ INFO] [1593268915.553417151]: HP: requesting home position
[ WARN] [1593268919.560201295]: CMD: Command 520 -- wait ack timeout
[ WARN] [1593268919.560751968]: PR: request list timeout, retries left 2
[ WARN] [1593268919.560987780]: VER: unicast request timeout, retries left 1
[ WARN] [1593268920.556941833]: CMD: Command 410 -- wait ack timeout
[ WARN] [1593268920.561007286]: PR: request list timeout, retries left 1
[ WARN] [1593268921.558353678]: WP: timeout, retries left 2
[ WARN] [1593268921.561846591]: PR: request list timeout, retries left 0
[ WARN] [1593268922.559670102]: WP: timeout, retries left 1
[ WARN] [1593268923.560162676]: WP: timeout, retries left 0
[ERROR] [1593268924.561451728]: WP: timed out.
[ WARN] [1593268924.564051079]: CMD: Command 520 -- wait ack timeout
[ WARN] [1593268924.564434707]: VER: unicast request timeout, retries left 0
[ INFO] [1593268925.553258953]: HP: requesting home position
[ WARN] [1593268929.568573596]: CMD: Command 520 -- wait ack timeout
[ WARN] [1593268929.569000617]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ WARN] [1593268930.557696087]: CMD: Command 410 -- wait ack timeout
[ INFO] [1593268935.553244725]: HP: requesting home position
[ WARN] [1593268940.558100571]: CMD: Command 410 -- wait ack timeout
[ INFO] [1593268945.553709390]: HP: requesting home position
[ WARN] [1593268950.557819343]: CMD: Command 410 -- wait ack timeout
[ INFO] [1593268955.553543099]: HP: requesting home position

I'll try to switch to a previous version of mavros, to identify if that's a possible issue.

Thanks again for your thoughts

rsisto, 2020-07-13:
Hi @vooon , sorry for the late reply but I finally found time to test this thoroughly. Starting, I tried changing mavros source as you suggested and the problem persists. 

After further tests, I used source install with these versions and everything works again.
mavros: release 0.32.0 
mavlink: release 2020.2.2-1

After this worked ok, I tried using mavros 0.33.0, which started failing with the original error.
I see there were a lot of changes in how messages are parsed, so something regarding fcu protocol v1.0 was broken (since it's legacy and there weren't probably enough tests).

I'll move forward with the versions stated above and moving on I'll update the hardware with a newer one.
I was hoping to leave this information in case some old school mavros user comes by :)

vooon, 2020-07-13:
Maybe you can switch to mavlink v2? As i remember it is even possible with APM 2.6.
So anything with STM32 should be fine.

debanik123, 2020-08-25:
I am having the same problem with apm 2.8 and mavros. Any solution would be appreciated?

debanik123, 2020-08-25:
Which mavros version is compatible with apm 2.8 ?

rsisto, 2020-08-26:
Hi @debanik123 , glad to see some activity here :)
Did you try downgrading mavros and mavlink versions? I made sure to add ways to determine the issue in this thread.

debanik123, 2020-08-26:
Yes I did according to your version

debanik123, 2020-08-26:
Can you please tell me which mavros version is working with apm2.8

debanik123, 2020-08-26:
@rsisto thanks for your replay

Nabil-mufc, 2022-08-13:
Hello all, so what are the best versions of ROS, MAVROS and MAVLINK to connect ROS to APM 2.8 through Rpi3B?

Nikhiljohnson21, 2023-08-31:
i also need help with this 
I'm trying to send commands via mavros to my pixhawk flight controller to control the drone. For now I'm testing with 1 motor.

Mavros is connected to pixhawk and I'm able to arm the motor
"rosrun mavros mavsafety arm"

Now I'm trying to take-off the drone using the command
"rosrun mavros mavcmd takeoff (with the yaw, pitch roll, latitude, longitude)"

But I'm getting "request failed. Check mavros logs. ACK:4"

I followed the steps instructed in mavros/utility commands documentation.

Coder-X15, 2025-05-10:
> This is only bug and feature tracker, please use it to report bugs or request features.
>
> ### Issue details
> Hi, I'm having issues connecting using mavros to ardupilot rover. Using APM Planner the [...]

I guess it's just the protocol version switch that needs a click. Set an argument `fcu_protocol:=v1.0`. It worked for me.

closed 2026-11-09 by github-actions[bot]