mavlink/mavros #1424: error: ‘BODY_FRD’ is not a member of ‘mavlink::common::MAV_FRAME’

cesarecaputi, 2020-05-07:
Hi,
when building the ~/catkin_ws I get this error message 

/catkin_ws/src/mavros/mavros_extras/src/plugins/odom.cpp:228:53: error: ‘BODY_FRD’ is not a member of ‘mavlink::common::MAV_FRAME’
   msg.child_frame_id = utils::enum_value(MAV_FRAME::BODY_FRD);

Does anybody know how to fix it?

vooon, 2020-05-07:
Update mavlink package.

vooon, 2020-05-07:
Well, seems that is opposite case. Too mavlink release 2020.5.5 breaks beacause of https://github.com/mavlink/mavlink/pull/1112.

I think temporary solution to replace enum_value with hardcoded value until it would be fixed in upstream.

vooon, 2020-05-07:
I added quick fix to master, please update.
https://github.com/mavlink/mavros/commit/1a898dea2202ee1af56d698bd40d40994346c5cb

Cristian-wp, 2020-05-07:
Thanks vooon, yesterday I was getting crazy about this error!

vooon, 2020-05-08:
Closing this as resolved. But feel free to reopen if needed.

closed 2020-05-08 by vooon

Cristian-wp, 2020-05-08:
Hi, I start a new project few minutes ago and the problem is still there...(I have fix it like you do).
I have follow the instruction from here
https://github.com/mavlink/mavros/tree/master/mavros#installation
maybe there is a problem with the master version(not updated with your changes)?

vooon, 2020-05-08:
@Cristian-wp commit was to master. Try to `git pull`.

hamishwillee, 2020-05-10:
@vooon YOu can perhaps help me. The above appears to be caused by a mavlink removal of a frame, which I think may have been incorrect. Can you help me understand whether that removal was appropriate? See https://github.com/mavlink/mavlink/pull/1112#issuecomment-625050092

vooon, 2020-05-12:
@hamishwillee unfortunately i'm not quire sure, my commit just a quick fix to solve compilation problem.
As i remember BODY_NED was actually FRD, so it's introducing reduce confusion a bit.

After https://github.com/mavlink/mavros/commit/1789e810d59cdad70ac0ada134254c96b05a06ed#diff-750c83d1bf274515183c544f142bac77 odom only uses FRD, not FLU.

hamishwillee, 2020-05-20:
Hi @vooon FYI, MAV_FRAME_BODY_FRD  just got added back into common.xml :-)

vooon, 2020-05-21:
@hamishwillee thanks! I'll release packages soon.

vooon, 2020-05-21:
Packages:
- https://github.com/ros/rosdistro/pull/24987
- https://github.com/ros/rosdistro/pull/24988
- https://github.com/ros/rosdistro/pull/24989
-

canberkgurel, 2020-05-26:
I solved this problem as follows:
1. Uninstall the mavlink: 

`sudo apt-get remove ros-kinetic-mavlink`

2. Clone the mavlink-gbp-release:

`cd catkin_ws/src/`
`git clone https://github.com/mavlink/mavlink-gbp-release.git`
`cd mavlink-gbp-release`
`git checkout release/kinetic/mavlink`

3. Build the catkin workspace:

`cd ~/catkin_ws`
`catkin clean -y`
`catkin build`