mavlink/mavros #1425: Can not read Pixhawk IMU data with mavros

Cristian-wp, 2020-05-07:
### Issue details

Hi, I am new to Mavros and ROS so please be patient :)

I am trying to read IMU linear acceleration data directly from a Pixhawk (connected via USB) with a C++ program without success.
Initially I had try to send command using this example:

> https://dev.px4.io/v1.9.0/en/ros/mavros_offboard.html

and in the terminal where I launch px4.launch I see the arming and offboard configuration without a problem. 

So I move forward and using a publisher and a subscriber I have tried to read the IMU (without setting the arming and the offboard like previously), but the value printed are all 0.

```
[ INFO] [1588859687.955951158]: 
msg_imu.linear_acceleration.z= 0.000000 
 msg.z= 0.000000 

[ INFO] [1588859688.019932463]: 
 Orientation X_list: 0.000000 
 Orientation Y_lis: 0.000000 
 Orientation Z_lis: 0.000000 
```

I have tried to see the values with rostopic echo but I can not see anything, the same thing happen with all other topic that I try. 
I think there is a problem of communication between my program and the PX4, but I do not find any clue...

This is the publisher:

```
#include "ros/ros.h"
#include "geometry_msgs/Vector3.h"
#include "sensor_msgs/Imu.h" //Imu data, orientation computed by FCU
#include "mavros_msgs/State.h"
#include "mavros_msgs/StreamRate.h"

sensor_msgs::Imu msg_imu; 
geometry_msgs::Vector3 msg;
mavros_msgs::State current_state;
mavros_msgs::StreamRate streamRate;

void state_cb(const mavros_msgs::State::ConstPtr& msg){
    current_state = *msg;
}

int main(int argc, char **argv){
    ros::init(argc, argv, "imu_data_talker");
    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>("mavros/state", 10, state_cb);
    ros::Publisher chatter_pub = nh.advertise<sensor_msgs::Imu>("/mavros/imu/data", 1000);
    ros::ServiceClient stream_rate_client = nh.serviceClient<mavros_msgs::StreamRate>("mavros/set_stream_rate");
    ros::Rate rate(15);   //the setpoint publishing rate MUST be faster than 2Hz

    // wait for FCU connection
    while(ros::ok() && !current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }
    ROS_INFO("Connection OK");

    streamRate.request.stream_id = 0;
    streamRate.request.message_rate = 100;
    streamRate.request.on_off = 1;
    if (true == stream_rate_client.call(streamRate))
    {
        ROS_INFO("OK. Stream rate set");
    }
    else
    {
[... 519 lines cut ...]
        value: "1.148785"
      - 
        key: "Mean RTT (ms)"
        value: "0.842208"
      - 
        key: "Last remote time (s)"
        value: "9643.207305000"
      - 
        key: "Estimated time offset (s)"
        value: "1588848785.971028805"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 1046 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 44, 77, 111, 245, 22

```

vooon, 2020-05-08:
I don't really know what is the problem, but i do not like that diag:
```
        key: "AHRS subsystem health"
        value: "Fail"
```

Probably you should calibrate the sensors. Also please check that attitude indicator in QGC works.

Cristian-wp, 2020-05-08:
I resolve it, I have not set the correct frame in QGC. Previously I had do some experiments with HITL and the UAV frame was not changed.

`
key: "AHRS subsystem health"
        value: "Fail"
`
Why you do not like that result? At the moment I have not armed the PX4 so I was thinking that was  normal...There is a software/hardware problem?

vooon, 2020-05-08:
I do not like that because AHRS is essential to be able to fly. You simply cannot stabilize your drone if you do not know reference orientation.
If i remember correctly on PX4 that refers to EKF health, but i have not use it for a while.

hichamhendy, 2021-07-19:
Could anyone guide me what I need to do configure the FCU? 

I also want to read IMU data from Pixhawk and echoing the topic /mavros/imu/data works fine by me but while publishing the code sticks in:     
while(ros::ok() && !current_state.connected){
        ros::spinOnce();
        rate.sleep();
        ROS_INFO("I am wating");
    }

I am not sure from the necessary configurations in general to solve the problem.
I will be grate for for the advice

hichamhendy, 2021-07-19:
I need some guidance by subscribing the topic /mavros/imu/data.

I also subscribed the topic but I get no response back even though echoing the topic works fine! 

I think I have a a problem with FCU, could anyone tell what I need to do configure this?

closed 2026-11-09 by github-actions[bot]