mavlink/mavros #1428: Mavlink HIGHRES_IMU data in mavros

Cristian-wp, 2020-05-13:
Hello,
I am trying to read the data in HIGHRES_IMU in ROS.
I need to get the pressure and the relative altitude computed in HIGHRES_IMU because I have set my correct values in QGroundControl in order to obtain a correct estimation of the altitude.
If I read the altitude from sensor_msgs::NavSatFix I have a huge error, is possible to set the sea level value in order to correct the error?
Otherwise is possible to get the values directly from QGC?
Thanks in advance.

vooon, 2020-05-14:
You may extend imu plugin: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/imu.cpp#L368
Or write your own handler for HIGHRES_IMU. It is ok to have several subscribers for the message.

Cristian-wp, 2020-05-15:
> You may extend imu plugin: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/imu.cpp#L368
> Or write your own handler for HIGHRES_IMU. It is ok to have several subscribers for the message.
Hi @vooon thank you for the fast responce.
I had a look at your imu.cpp but I dont understad how to get mavlink data inside a ros custom package. For example I  create a Pressure_pkg, how can I read the data with mavlink?
I have already read the developer guide, but I haven't understand how to procede...
Can you give me a hand please?

Cristian-wp, 2020-05-15:
I have try another way but same results...
I have create another package for the IMU in order to read the following topics:
`/mavros/imu/atm_pressure`
`/mavros/global_position/rel_alt`
`/mavros/imu/data`
I really don't understand why but I get only the IMU data, the other topics are not launched (checked with rqt_graph).
Can you tell me why?

This is my code:
------------------------------------------------------------------------------
#include "ros/ros.h"
#include "geometry_msgs/Vector3.h"

#include "mavros_msgs/State.h"                          // FCU state - ~state
                                                        

#include "std_msgs/Float64.h"                             // Relative altitude - ~global_position/rel_alt
                                                        // Compass heading in degrees - ~global_position/compass_hdg
#include "sensor_msgs/Temperature.h"
#include "sensor_msgs/Imu.h"
#include "sensor_msgs/FluidPressure.h"

mavros_msgs::State current_state;
void state_cb(const mavros_msgs::State::ConstPtr &msg){
    current_state = *msg;
}
```

void chatterBAROCallback(const sensor_msgs::FluidPressure::ConstPtr &msgBaro)
{
   ROS_INFO("\n BARO-id: %i \n BARO-height[m]: %f \n \n\n",msgBaro->header.seq, msgBaro->fluid_pressure);
}

void chatterGlobPosCallback(const std_msgs::Float64::ConstPtr &msgGlobPos)
{
   ROS_INFO("\n Relative altitude[m]: %f \n \n\n",msgGlobPos->data );
}

void chatterIMUCallback(const sensor_msgs::Imu::ConstPtr& msgIMU)
{
  ROS_INFO("\n Orientation X_list: %f \n Orientation Y_lis: %f \n Orientation Z_lis: %f \n\n", msgIMU->linear_acceleration.x,msgIMU->linear_acceleration.y,msgIMU->linear_acceleration.z);
}

int main(int argc, char **argv){

    ros::init(argc, argv, "imu_node");

    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>("mavros/state", 10, state_cb);
    ros::Subscriber imu_sub = nh.subscribe("/mavros/imu/data", 1000,chatterIMUCallback);
    ros::Subscriber baro_sub = nh.subscribe<sensor_msgs::FluidPressure>("mavros/imu/atm_pressure", 1000,chatterBAROCallback);
    ros::Subscriber GlobPos_sub =nh.subscribe<std_msgs::Float64>("mavros/global_position/rel_alt", 1000,chatterGlobPosCallback);
    
    ros::Rate rate(20); //Hz

    // wait for FCU connection
    while(ros::ok() && !current_state.connected){
        ros::spinOnce();
        rate.sleep();
        if(current_state.connected){
            //ROS_INFO(" FCU Connected \n");    
        }
    }
    while(ros::ok() && current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }
    return 0;
}
```

------------------------------------------------------------------------------
FCU version: PX4 v1.10
 I work on Ubuntu 18.04 with ROS Melodic

closed 2026-11-08 by github-actions[bot]