mavlink/mavros #1429: mavros_extras fails to build on Kinetic

tfoote, 2020-05-13:
The latest release of mavros_extras is failing to build on Kinetic due to a missing symbol in mavlink. 

It aeears to have been triggered by the release of mavlink 2020.5.5
[image]

`16:10:40 /tmp/binarydeb/ros-kinetic-mavros-extras-1.1.0/src/plugins/odom.cpp:228:42: error: ‘BODY_FRD’ is not a member of ‘mavlink::common::MAV_FRAME’
16:10:40    msg.child_frame_id = utils::enum_value(MAV_FRAME::BODY_FRD);`

http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__mavros_extras__ubuntu_xenial_amd64__binary/29/console

<details>

```
16:10:40 /tmp/binarydeb/ros-kinetic-mavros-extras-1.1.0/src/plugins/odom.cpp: In member function ‘void mavros::extra_plugins::OdometryPlugin::odom_cb(const ConstPtr&)’:
16:10:40 /tmp/binarydeb/ros-kinetic-mavros-extras-1.1.0/src/plugins/odom.cpp:228:42: error: ‘BODY_FRD’ is not a member of ‘mavlink::common::MAV_FRAME’
16:10:40    msg.child_frame_id = utils::enum_value(MAV_FRAME::BODY_FRD);
16:10:40                                           ^
16:10:40 CMakeFiles/mavros_extras.dir/build.make:353: recipe for target 'CMakeFiles/mavros_extras.dir/src/plugins/odom.cpp.o' failed
16:10:40 make[4]: Leaving directory '/tmp/binarydeb/ros-kinetic-mavros-extras-1.1.0/obj-x86_64-linux-gnu'
16:10:40 make[4]: *** [CMakeFiles/mavros_extras.dir/src/plugins/odom.cpp.o] Error 1
16:10:40 make[3]: *** [CMakeFiles/mavros_extras.dir/all] Error 2
16:10:40 CMakeFiles/Makefile2:186: recipe for target 'CMakeFiles/mavros_extras.dir/all' failed
16:10:40 make[3]: Leaving directory '/tmp/binarydeb/ros-kinetic-mavros-extras-1.1.0/obj-x86_64-linux-gnu'
16:10:40 Makefile:130: recipe for target 'all' failed
16:10:40 make[2]: *** [all] Error 2
16:10:40 make[2]: Leaving directory '/tmp/binarydeb/ros-kinetic-mavros-extras-1.1.0/obj-x86_64-linux-gnu'
16:10:40 dh_auto_build: make -j1 returned exit code 2
16:10:40 	cd /tmp/binarydeb/ros-kinetic-mavros-extras-1.1.0
16:10:40 make[1]: *** [override_dh_auto_build] Error 2
16:10:40 debian/rules:35: recipe for target 'override_dh_auto_build' failed
16:10:40 make[1]: Leaving directory '/tmp/binarydeb/ros-kinetic-mavros-extras-1.1.0'
16:10:40 debian/rules:22: recipe for target 'build' failed
16:10:40 make: *** [build] Error 2
16:10:40 dpkg-buildpackage: error: debian/rules build gave error exit status 2
16:10:40 E: Building failed
16:10:40 Traceback (most recent call last):
16:10:40   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 138, in build_binarydeb
16:10:40     subprocess.check_call(cmd, cwd=source_dir)
16:10:40   File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
16:10:40     raise CalledProcessError(retcode, cmd)
16:10:40 subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-kinetic-mavros-extras']' returned non-zero exit status 1
16:10:40 
```
</details>

Jaeyoung-Lim, 2020-05-13:
This is not limited to kinetic, and was introduced by https://github.com/mavlink/mavlink/pull/1112

vooon, 2020-05-14:
@tfoote i have a fix in master (a bit dirty, but that should give some time).
And i'm going to release it, but first want to have high-precision mission support merged: https://github.com/mavlink/mavros/pull/1415

vooon, 2020-05-22:
Should be fixed now.

closed 2020-05-22 by vooon