mavlink/mavros #1435: Catkin Build, Mavros Fails

Reuben-Isaac, 2020-05-30:
Hi everyone. I recently installed mavros/mavlink through the setup script provided by PX4 [https://raw.githubusercontent.com/PX4/Devguide/v1.10/build_scripts/ubuntu_sim_ros_melodic.sh](url). After this, I ran `source ~/catkin_ws/devel/setup.bash`. This adds the mavlink and mavros packages to the ROS_PACKAGE_PATH. The issue comes when I try `rosrun mavros mavros_node`. The command line says "Couldn't find an executable named mavros_node below /home/undergrad/catkin_ws/src/mavros/mavros" The node is clearly in this folder but it cannot find it. I decided to rerun `catkin build` and found this report
[image]
I believe the previous issues have something to do with the errors shown in the image above. I am somewhat new to ROS and Linux and all of this stuff so if anyone could decode the error message and guide me through what needs to be done that would be helpful. Thank you all.

Reuben-Isaac, 2020-05-30:
Hi guys, just figured out that running these commands solve the issue.
`cd /your/catkin/workspace/root`
`rosdep install --from-paths src --ignore-src`
This automatically installs all required binary dependencies for the packages in your workspace.
Source: ayrton04 [https://github.com/cra-ros-pkg/robot_localization/issues/245](url)

closed 2020-05-30 by Reuben-Isaac

TSC21, 2020-05-31:
@Reuben-Isaac your first source should be MAVROS instructions: https://github.com/mavlink/mavros/tree/master/mavros#source-installation