mavlink/mavros #1437: Mavros 1.2 not working properly URGENT

bavokn, 2020-06-02:
----

### Issue details

I'm working with docker and I am using ros-melodic-mavros and ros-melodic-mavros-extras for controlling a PX4 iris drone. in the version 1.1 it worked just fine. I am controlling drones using Positiontarget  to the topic : '/iris_{}/mavros/setpoint_raw/local but nothing happens in 1.2. There is no option to getting a previous version from apt-get (why ?)

### MAVROS version and platform

Mavros: 1.2
ROS: melodic
Ubuntu:18.04
docker : 19.03.11
Gazebo : 9

### Autopilot type and version

[  ] ArduPilot
[ x  ] PX4

Version: 1.10

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
header: 
  seq: 889
  stamp: 
    secs: 1345
    nsecs: 601000000
  frame_id: ''
status: 
  - 
    level: 0
    name: "iris_1/mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Received packets:"
        value: "27630"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "238"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "24864594"
      - 
        key: "Tx total bytes:"
        value: "735053"
      - 
        key: "Rx speed:"
        value: "24529.000000"
      - 
        key: "Tx speed:"
        value: "742.000000"
  - 
    level: 0
    name: "iris_1/mavros: GPS"
    message: "3D fix"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Satellites visible"
        value: "10"
      - 
        key: "Fix type"
        value: "3"
      - 
        key: "EPH (m)"
        value: "0.00"
      - 
        key: "EPV (m)"
        value: "0.00"
  - 
    level: 0
    name: "iris_1/mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "1016"
      - 
        key: "Frequency (Hz)"
        value: "0.951928"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "OFFBOARD"
      - 
        key: "System status"
        value: "Active"
  - 
    level: 2
    name: "iris_1/mavros: System"
    message: "Sensor health"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Sensor present"
        value: "0x1020002F"
      - 
        key: "Sensor enabled"
        value: "0x1021000F"
      - 
        key: "Sensor health"
        value: "0x1020002F"
      - 
        key: "3D gyro"
        value: "Ok"
      - 
        key: "3D accelerometer"
        value: "Ok"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "rc receiver"
        value: "Fail"
      - 
        key: "AHRS subsystem health"
        value: "Ok"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "iris_1/mavros: Battery"
    message: "Normal"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Voltage"
        value: "11.32"
      - 
        key: "Current"
        value: "-1.0"
      - 
        key: "Remaining"
        value: "51.0"
  - 
    level: 0
    name: "iris_1/mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://:14541@localhost:14581"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "10168"
      - 
        key: "Frequency (Hz)"
        value: "9.995240"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.117328"
      - 
        key: "Last remote time (s)"
        value: "1016.949000000"
      - 
        key: "Estimated time offset (s)"
        value: "328.622159175"
---

```

### Check ID

```
WARNING: mavros/target_system_id not set. Used default value: 1
WARNING: mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

###Check ID of single drone namespace 
rosrun mavros checkid --mavros-ns iris_1/mavros

```
ERROR. I got 0 addresses, but not your target 2:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages
``` 

Points are being published (checked with rostopic echo) but the drone is not moving at all.

VicSegers, 2020-06-02:
I am having the same problem. My code worked in 1.1.0, after being forced to upgrade to 1.2.0 everything broke. Now my UAV has problems with its gyroscope for some odd reason.

vooon, 2020-06-03:
You can checkout 1.1.0 and build it from source. But i think that there something else because setpoint_raw stay the same for years.

closed 2026-09-21 by bavokn