mavlink/mavros #1444: Documentation: wrong topic in external vision odometry

marcelino-pensa, 2020-06-20:
### Issue details
I am using Px4 1.10 with Mavros 1.2.0. I was following the documentation [from here](https://dev.px4.io/v1.10/en/ros/external_position_estimation.html) to feed odometry to px4. It took me a while to realize that the documentation was wrong, as I have to publish into the topic `/mavros/odometry/out` instead of the documented one (`/mavros/odometry/odom`):

[image]

We can find the correct topics in `mavros_extras/src/plugins/odom.cpp`.

### MAVROS version and platform

Mavros: 1.2.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: 1.10.0

TSC21, 2020-06-20:
Please bring this to the PX4 devguide issue or propose a PR to fix it there. Thank you!

closed 2020-06-20 by TSC21

hamishwillee, 2020-06-21:
@marcelino-pensa FYI this is easy
1. Go to page: https://dev.px4.io/master/en/ros/external_position_estimation.html
2. Up the top of the page there are buttons for editing the page (creating PR) on github or creating an issue.
   [image]

For cases like this it is much better to just create a PR.

Note, this will create the fix in the master branch. Normally we don't back-fix old versions.

marcelino-pensa, 2020-06-23:
Thanks @hamishwillee , I have just created an issue and a PR for that!