mavlink/mavros #1445: line 27: hash: geographiclib-get-geoids: not found

lkw303, 2020-06-21:
### Issue details
Hi I'm not sure why the output below is shown everytime  I run ```sudo ~/drone_ws/src/mavros/mavros/scripts/install_geographiclib_datasets.sh```

### MAVROS version and platform

Mavros: 1.2.0
ROS: Melodic
Ubuntu: 18.04

```
line 27: hash: geographiclib-get-geoids: not found
/home/kai/drone_ws/src/mavros/mavros/scripts/install_geographiclib_datasets.sh: line 31: hash: geographiclib-datasets-download: not found
OS not supported! Check GeographicLib page for supported OS and lib versions.

```

lkw303, 2020-06-23:
Created a new workspace and loaded the all the packages again and it worked. Not exactly sure why.

closed 2020-06-23 by lkw303

SEUZTh, 2020-11-21:
Hello. How did you install GeographicLib on Ubuntu18.04? It seems that it only supports Ubuntu16.04