mavlink/mavros #1448: /mavros/local_position/local topic is publishing nothing?

aircraftpilot5, 2020-07-10:
I am using Erle-Brain. "/mavros/local_position/local" topic is publishing nothing. (No handlers could be found for logger "rosout") There is not any problem with the other topics. How i figure out this problem? Could anyone help about this issue? 

pi@erle-brain-2 ~ $ rostopic list
No handlers could be found for logger "rosout"
/camera/camera_info
/camera/image/compressed
/diagnostics
/mavlink/from
/mavlink/to
/mavros/actuator_control
/mavros/battery
/mavros/cam_imu_sync/cam_imu_stamp
/mavros/extended_state
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/local
/mavros/global_position/raw/fix
/mavros/global_position/raw/gps_vel
/mavros/global_position/rel_alt
/mavros/image/camera_image
/mavros/image/camera_image/compressed
/mavros/image/camera_image/compressed/parameter_descriptions
/mavros/image/camera_image/compressed/parameter_updates
/mavros/imu/atm_pressure
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/mag
/mavros/imu/temperature
/mavros/local_position/local
/mavros/manual_control/control
/mavros/mission/waypoints
/mavros/mocap/pose
/mavros/px4flow/ground_distance
/mavros/px4flow/raw/optical_flow_rad
/mavros/px4flow/temperature
/mavros/radio_status
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/safety_area/set
/mavros/setpoint_accel/accel
/mavros/setpoint_attitude/att_throttle
/mavros/setpoint_attitude/attitude
/mavros/setpoint_attitude/cmd_vel
/mavros/setpoint_position/local
/mavros/setpoint_velocity/cmd_vel
/mavros/state
/mavros/time_reference
/mavros/vfr_hud
/mavros/vibration/raw/vibration
/mavros/vision_pose/pose
/mavros/vision_pose/pose_cov
/mavros/vision_speed/speed_vector
/mavros/wind_estimation
/rosout
/rosout_agg
/tf
/tf_static

### MAVROS version and platform

Mavros: ?
ROS: indigo
Raspbian GNU/Linux 8.0 (jessie)

### Autopilot type and version

[ x] ArduPilot
[  ] PX4

### Diagnostics

```
header: 
  seq: 2691
  stamp: 
    secs: 1594384472
    nsecs: 735012469
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: udp://:6001@
    values: 
      - 
        key: Received packets:
        value: 55505
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 80
      - 
        key: Tx sequence number:
        value: 86
      - 
        key: Rx total bytes:
        value: 8644260
      - 
        key: Tx total bytes:
        value: 9522
      - 
        key: Rx speed:
        value: 3597.000000
      - 
        key: Tx speed:
        value: 0.000000
  - 
    level: 0
    name: mavros: GCS bridge
    message: connected
    hardware_id: udp://:6001@
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 0
      - 
        key: Rx total bytes:
        value: 0
      - 
        key: Tx total bytes:
        value: 7351715
      - 
        key: Rx speed:
        value: 0.000000
      - 
        key: Tx speed:
        value: 3079.000000
  - 
    level: 0
    name: mavros: GPS
    message: 3D fix
    hardware_id: udp://:6001@
    values: 
      - 
        key: Satellites visible
        value: 8
      - 
        key: Fix type
        value: 4
      - 
        key: EPH (m)
        value: 1.17
      - 
        key: EPV (m)
        value: 1.87
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: udp://:6001@
    values: 
      - 
        key: Heartbeats since startup
        value: 3524
      - 
        key: Frequency (Hz)
        value: 1.000023
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: ArduPilotMega
      - 
        key: Mode
        value: GUIDED
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:6001@
    values: 
      - 
        key: Sensor present
        value: 0x0060FC2F
      - 
        key: Sensor enabled
        value: 0x0060FC2F
      - 
        key: Sensor helth
        value: 0x0060FC2F
      - 
        key: Sensor 3D Gyro
        value: Ok
      - 
        key: Sensor 3D Accel
        value: Ok
      - 
        key: Sensor 3D Mag
        value: Ok
      - 
        key: Sensor Abs Pressure
        value: Ok
      - 
        key: Sensor GPS
        value: Ok
      - 
        key: Sensor Ang Rate Control
        value: Ok
      - 
        key: Sensor Attitude Stab
        value: Ok
      - 
        key: Sensor Yaw Position
        value: Ok
      - 
        key: Sensor Z/Alt Control
        value: Ok
      - 
        key: Sensor X/Y Pos Control
        value: Ok
      - 
        key: Sensor Motor Outputs
        value: Ok
      - 
        key: AHRS health
        value: Ok
      - 
        key: Terrain health
        value: Ok
      - 
        key: CPU Load (%)
        value: 75.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: udp://:6001@
    values: 
      - 
        key: Voltage
        value: 11.78
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 100.0
  - 
    level: 0
    name: mavros: 3DR Radio
    message: Normal
    hardware_id: udp://:6001@
    values: 
      - 
        key: RSSI
        value: 154
      - 
        key: RSSI (dBm)
        value: -45.9
      - 
        key: Remote RSSI
        value: 152
      - 
        key: Remote RSSI (dBm)
        value: -47.0
      - 
        key: Tx buffer (%)
        value: 83
      - 
        key: Noice level
        value: 86
      - 
        key: Remote noice level
        value: 60
      - 
        key: Rx errors
        value: 5
      - 
        key: Fixed
        value: 0

berkeyvx, 2020-09-30:
Before trying to get data from a topic, set stream rate using this command

rosservice call /mavros/set_stream_rate 0 10 1

After call

rostopic echo /mavros/local_position/local

koi-coder, 2023-05-16:
> Before trying to get data from a topic, set stream rate using this command
> 
> rosservice call /mavros/set_stream_rate 0 10 1
> 
> After call
> 
> rostopic echo /mavros/local_position/local

I just want to say thank you a lot. You really saved me.

closed 2026-11-08 by github-actions[bot]