mavlink/mavros #1449: Mavros command services

HendAdell, 2020-07-11:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[x] ArduPilot
[  ] PX4

I have a question can I use command.takeoff(), command.land() and command.arm() from mavros, without making a ros node, but instead using a normal python code ? if yes, how is that possible and why ?

rezaqoli, 2020-07-13:
hey, try this:
`rosrun mavros mavcmd takeoff`
for further information about how this work , you can look at this directiry---->MAVROS/mavros/scripts

HendAdell, 2020-07-13:
I already have a python code which is not a ros node, it uses MAVROS services, such as arming, taking off and landing. I was wondering how was that possible.
I imported from mavros, command

vooon, 2020-07-13:
@HendAdell your code effectively is a node. It imports rospy.

HendAdell, 2020-07-13:
It is not affected with importing rospy, If I removed import rospy, the code still works effectively. However, it is only affected by importing mavros
Also when I run in a terminal rosrun rqt_graph rqt_graph, it doesn't appear as a node.

vooon, 2020-07-14:
1. https://github.com/mavlink/mavros/blob/master/mavros/src/mavros/command.py
2. anonymous node. use console tools.

HendAdell, 2020-07-15:
I checked in the terminal with rosnode list and it just gave me /mavros  and /rosout nodes only, no annonymous nodes.

closed 2026-11-08 by github-actions[bot]