mavlink/mavros #1450: mavros connecting error between Jetson Nano and Pixhawk

mzwtju, 2020-07-13:
----

### Issue details

Hi, I'm using Jetson Nano as a companion computer for Pixhawk . 

I followed the instruction [https://dev.px4.io/master/en/companion_computer/pixhawk_companion.html#companion-computer-for-pixhawk-series](url)

I made the same configuration as I did on rasberrypi which works fine, but I got the following warnings and errors.  I can't use offboard mode at all. 

Can anyone tell me what's wrong in my logs and diagnostic? Or what's the probable cause to these errors and warnings. I am new to this.

Thanks in advance.

### MAVROS version and platform

Mavros: 1.2.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[  ] PX4

Version: 1.10.1

### Node logs

```
... logging to /home/amov/.ros/log/dc5a537e-c50e-11ea-9339-d6828bb119a5/roslaunch-amov-12836.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://amov:41377/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
[... 565 lines cut ...]
 35:6     84737
 111:9     62
 213:6     84737
  1:75    14417921, 1245185
 107:6     84737
  1:31    11403264, 7536640
  0:1     32, 30, 31
 162:6     7681, 84737
 35:15    187
 35:93    183
 191:32    51
  6:1     0, 1, 2, 4, 3080222, 140, 141, 16, 323, 24, 14303748, 15204382, 31, 32, 36, 7405599, 7581983, 30, 8519711, 4784160, 331, 74, 65611, 76, 14221453, 75, 340, 4915230, 12293924, 6785566, 335, 97, 230, 105, 111, 241, 147, 245, 254, 255
  6:31    6160384
 213:109   12
 215:167   185
  0:75    9633793
  0:64    4915462
 148:6     84737
 67:6     7681
 28:0     0
  1:74    0
  1:30    7536640, 11730944, 2752512, 4259840, 9371648
 71:6     84737

```

vooon, 2020-07-13:
Hello,

Check IDs result looks unusual to me. Did you connect directly FCU and Jetson?
That high message ids are too big.
Either you have very noisy uart connection, or you have a problem with software.
Warnings from param plugin also indicates that some of the messages lost.

mzwtju, 2020-07-14:
@vooon  thanks for your comment.
YES, I directly connect FCU Telem2 port and Jetson Nano UART port with wires.

vooon, 2020-07-14:
1. Check your wiring, and that they away from power wired to motors
2. Try to use something like ser2net and QGC to see if it works

mzwtju, 2020-07-17:
@vooon  thanks for your suggestion but it didn't work.
Probably there's something wrong with Linux environment setup on Jetson Nano . I will try to re-install system later.
I will update when I fix my problem

closed 2020-07-17 by mzwtju

vooon, 2020-07-17:
@mzwtju to check wiring problem also possible to use USB, just to be sure that this isn't a software problem.
USB has it's own additional protection against transmission errors.

Your results a bit confusing, because mavlink itself have crc16 protection, which is more than enough for 256-byte long frames.
Potentially there may be a problem in mine c++11 mavlink decoder part, but i don't seen errors before.
And it's worked fine on raspberry, so probably source, unaligned memory access is not the case.
But newer compillers complains about it.

mzwtju, 2020-07-18:
@vooon  Appreciate your help! I think I've found the reason why Jetson Nano didn't work well.
I used USB to connected to PX4 as you mentioned, it worked without missing param.
I made a stupid mistake that GND pins were not connect!!!!
Thank you again!

6amarj, 2022-08-05:
Hi, 

@mzwtju  just wanted to ask a question regarding what you've done. I'm trying to do something similar.....right now my jetson is connected through Telem2 and my terabee lidar is connected to the I2C. However, when i launch mavros and try to look for the lidar output in /mavros/distance_sensor/rangefinder_pub topic, i get this......(image attached
[image]
)