mavlink/mavros #1452: Cannot set home in SITL Arducopter

gitdillo, 2020-07-22:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

I am trying to set home using a simulated Arducopter but I can not get it to work. I have tried the service as well as the mavcmd utility and both fail.

This is the result of calling the set_home service (note that I am using the /Klotho/mavros namespace):

```
$ rosservice call /Klotho/mavros/cmd/set_home "{current_gps: false, yaw: 0.0, latitude: 5.0, longitude: 1.0, altitude: 100.0}" 
success: False
result: 4
```

When trying the same but with "current_gps: true", it works but it is useless since I want to set my own coordinates, not use the current GPS ones:

```
$ rosservice call /Klotho/mavros/cmd/set_home "{current_gps: true, yaw: 0.0, latitude: 5.0, longitude: 1.0, altitude: 100.0}" 
success: True
result: 0
```

I get the same results when trying with the utility command:

```
$ rosrun mavros mavcmd -n Klotho/mavros sethome 10 10 100
Request failed. Check mavros logs. ACK: 4
```

The MAVLINK response in both cases is 4, which I think refers to the [MAV_RESULT_FAILED](https://mavlink.io/en/messages/common.html#MAV_RESULT_FAILED) mavlink message.

The mavros logs mentioned in the response are not particularly helpful, here is the output from the latest file in the ~/.ros/log directory:

```
[rospy.client][INFO] 2020-07-22 16:37:05,954: init_node, name[/mavcmd_92877_1595428625933], pid[92877]
[xmlrpc][INFO] 2020-07-22 16:37:05,955: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2020-07-22 16:37:05,955: Started XML-RPC server [http://nordluft-Inspiron-5370:39037/]
[rospy.impl.masterslave][INFO] 2020-07-22 16:37:05,955: _ready: http://nordluft-Inspiron-5370:39037/
[rospy.init][INFO] 2020-07-22 16:37:05,955: ROS Slave URI: [http://nordluft-Inspiron-5370:39037/]
[rospy.registration][INFO] 2020-07-22 16:37:05,955: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2020-07-22 16:37:05,956: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2020-07-22 16:37:06,056: registered with master
[rospy.rosout][INFO] 2020-07-22 16:37:06,056: initializing /rosout core topic
[rospy.rosout][INFO] 2020-07-22 16:37:06,067: connected to core topic /rosout
[rospy.simtime][INFO] 2020-07-22 16:37:06,071: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.core][INFO] 2020-07-22 16:37:06,094: signal_shutdown [atexit]
[rospy.impl.masterslave][INFO] 2020-07-22 16:37:06,099: atexit
mavcmd_92877_1595428625933.log (END)
```

### MAVROS version and platform

Mavros: Don't know how to get version, installed July 2020.
ROS: noetic
Ubuntu: 20.04

### Autopilot type and version

ArduPilot in SITL
Version: ArduCopter V4.1.0-dev (98feb061)

RamanaBotta, 2021-11-29:
Same issue 

rosservice call /mavros/cmd/set_home "{current_gps: true, yaw: 0.0, latitude: 13.589722, longitude: 80.003377, altitude: 8.0}" 
success: False
result: 1

vooon, 2021-11-29:
Without `FCU: foo bar` log lines i can only try fortune telling on coffee grounds.

sherlockholmes1603, 2022-06-12:
I am facing the same issue. Can anyone please guide me how to solve the issue

hfcaio, 2024-05-08:
I have the same problem. How did you solve ?

closed 2026-11-08 by github-actions[bot]