mavlink/mavros #1455: Changing mavros topic publishing rate

KonRoz, 2020-07-23:
### Issue details
I have been using message filters to construct a status message from several MAVROS topics. That said, one of the topics that I am using, battery, publishes very slow (~0.5 hz) and thus limits the speed at which I can publish state messages to 0.5 hz. I would like to change the frequency at which messages are published to the battery topic to speed up my message filter. I haven't been able to find an answer in the documentation. 

Is it possible to change the publishing rate? If so, does anyone know how to do this? 

Thanks

### MAVROS version and platform

Mavros: 1.2.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

amilcarlucas, 2020-10-09:
Just tell PX4 to publish that Mavlink  message more often, and mavros will then also publish the ROS messages faster

closed 2020-10-09 by amilcarlucas

aihamalharbat, 2020-11-13:
@amilcarlucas can you please provide me with any pointers to how we can "tell PX4 to publish that Mavlink message more often".

amilcarlucas, 2020-11-13:
Check the documentation of the SET_MESSAGE_INTERVAL mavlink message.

vooon, 2020-11-13:
@amilcarlucas also have a look at: https://docs.px4.io/master/en/peripherals/serial_configuration.html

mariofaragalla, 2021-04-19:
Well I have the problem but I am only doing gazebo simulation can anyone help???

tcies, 2021-11-23:
Seems to be part of the system status plugin: https://github.com/mavlink/mavros/blob/0a7df10a40291fd5c75dc191692e34290bc61744/mavros/src/plugins/sys_status.cpp#L517

tcies, 2021-11-23:
nvm, https://answers.ros.org/question/361371/increasechange-mavros-local_position-pose-publish-frequency/

tcies, 2021-11-23:
Check the mavlink id of the desired message, for example odometry:

https://mavlink.io/en/messages/common.html#ODOMETRY 

then, to set odometry to 100

```bash
rosservice call /mavros/set_message_interval 331 100
rostopic hz /mavros/odometry/in
```