mavlink/mavros #1457: Problem in catkin build

adhudhwani99, 2020-07-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I don't know why always mavros:make and mavros node is getting failed. I am very new to this so please help me

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[ . ] PX4 

Version: ?1.9.0?
mavros

```
Errors     << mavros:make /home/adhudhwani/catkin_ws/logs/mavros/build.make.001.log
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros_node.dir/src/mavros_node.cpp.o] Error 4
make[1]: *** [CMakeFiles/mavros_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/3dr_radio.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/actuator_control.cpp.o] Error 4
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
cd /home/adhudhwani/catkin_ws/build/mavros; catkin build --get-env mavros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

```

```
Failed     << mavros:make                  [ Exited with code 2 ]              
Failed    <<< mavros                       [ 6 minutes and 32.0 seconds ]      
Abandoned <<< mavros_extras                [ Unrelated job failed ]            
Abandoned <<< test_mavros                  [ Unrelated job failed ]            
[build] Summary: 3 of 6 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 2 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 6 minutes and 36.3 seconds total. 

```

### Check ID

```
rosrun mavros checkid
Traceback (most recent call last):
  File "/home/adhudhwani/catkin_ws/src/mavros/mavros/scripts/checkid", line 21, in <module>
    import mavros
ImportError: No module named mavros

```

vooon, 2020-07-27:
Hello, what is your environment? I mean what is the computer where you tried to build mavros?
Usually i see reports about `cc1plus killed` when people tries to compile on raspberry pies, with small ram.

Try to use following options:
1. use `catkin build -j1`, which will run only one instance of the g++, that slower, but you'll fit in 1 GiB RAM
2. attach a usb drive / flash and use it for swap file. Note: even usb 2.0 thumb drives faster than sd cards

vooon, 2020-07-27:
Additional note: placing sources to the flash drive also will help you build faster.

closed 2026-11-07 by github-actions[bot]