mavlink/mavros #1460: What is difference between velocity_local and velocity_body?

ijeong2125, 2020-08-03:
What is difference between /mavros/local_position/velocity_local and /mavros/local_position/velocity_body??

TSC21, 2020-08-03:
The first is the velocity in the local frame of reference, the second on the body frame. In ROS, the first is usually associated with the `map` or `odom` frames, while the second with the `base_link`. Check the [ROS REP 105](https://www.ros.org/reps/rep-0105.html) for me context.

closed 2020-08-03 by TSC21

ijeong2125, 2020-08-04:
Thank you!
I have other question.
What is difference between /mavros/setpoint_attitude/attitude and /mavros/setpoint_raw/attitude??

tejalbarnwal, 2023-08-24:
Hi @TSC21,
I am trying to understand the same difference.
I am using Ardupilot with ROS Noetic.
I am trying to compare the outputs from `/mavros/local_position/velocity_local`, `/mavros/local_position/velocity_body` and `/mavros/local_position/odom`.

I plotted some graphs to analyze the outputs from these messages.

With the graph attached, I concluded that `/mavros/local_position/velocity_body` and `/mavros/local_position/odom` represent the same values. 
[image]
[image]

Further, I also echoed these topics in order to observe the frames in which they are published. 
I found that `/mavros/local_position/velocity_local` has frame_id as `base_link`, `/mavros/local_position/odom` has frame_id as `map` and child_frame_id as `base_link`, finally `/mavros/local_position/velocity_body` has frame_id as `base_link`.

I think, as you mentioned above in the thread, the `/mavros/local_position/velocity_local` provides velocity in the local frame, so shouldn't the it have its frame_id as `map`?
therefore, could you please clarify the difference between these topics, especially `/mavros/local_position/velocity_local` and `/mavros/local_position/velocity_body`, and why there is this slight difference between the velocities in `map` frame and `base_link` frame?