mavlink/mavros #1461: Cannot set parameter to 0.0 using /mavros/param/set

marcelino-pensa, 2020-08-04:
### Issue details

I am running px4 1.10 along with mavros 1.2.0. I am trying to use the service `/mavros/param/set` to set a real-valued variable to `0.0`. The issue is that when I am trying to set it to `0.0`, `mavros` interprets as if I am trying to set an `integer` variable to `0`, and px4 returns the error `FCU: [pm] param types mismatch param`.
If I try to set the same variable to `0.000000001`, it works, but I had to dig into the code to figure out that this was the case.
The issue can be found in `mavros/mavros/src/plugins/param.cpp`, within the function `set_cb`. The snippet below makes it clear:

```
// according to ParamValue description
if (req.value.integer != 0)
	to_send.param_value = static_cast<int>(req.value.integer);
else if (req.value.real != 0.0)
	to_send.param_value = req.value.real;
else
	to_send.param_value = 0;
```

I think that the issue at hand is that we are using a single message type (`mavros_msgs::ParamSet`) to send both integers and real-values to px4. 

Possible solutions:
1) Create new message types `mavros_msgs::ParamSetReal` and `mavros_msgs::ParamSetInteger`, and use them accordingly with services as `/mavros/param/set_real` and `/mavros/param/set_integer`. We can still keep the message type `mavros_msgs::ParamSet` and the current ROS service so it doesn't break current functionality on people using mavros.
2) Create a new entry in `mavros_msgs::ParamSet` indicating that we want to set an integer or a real variable. Something like `is_integer`, which can be set to`true` or `false`. A disadvantage of this is that it might be a breaking change to those using mavros.

I'd be happy in contribute to the solution to this, but I'd like to hear feedback from you guys on what you think is the best solution to this problem!

### MAVROS version and platform

Mavros: 1.2.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: 1.10

### Node logs

```
[ INFO] [1596579271.517476079, 48.264000000]: FCU: [pm] param types mismatch param: EKF2_EV_POS_X
[ WARN] [1596579272.310265619, 49.056000000]: PR: Resend param set for EKF2_EV_POS_X, retries left 2
[ INFO] [1596579272.313466940, 49.060000000]: FCU: [pm] param types mismatch param: EKF2_EV_POS_X
[ INFO] [1596579273.254614588, 50.000000000]: HP: requesting home position
[ WARN] [1596579273.310988008, 50.056000000]: PR: Resend param set for EKF2_EV_POS_X, retries left 1
[ INFO] [1596579273.314756302, 50.060000000]: FCU: [pm] param types mismatch param: EKF2_EV_POS_X
[ WARN] [1596579274.311843994, 51.056000000]: PR: Resend param set for EKF2_EV_POS_X, retries left 0
[ INFO] [1596579274.315734815, 51.060000000]: FCU: [pm] param types mismatch param: EKF2_EV_POS_X
[ERROR] [1596579275.313010805, 52.056000000]: PR: Param set for EKF2_EV_POS_X timed out.

```

### Diagnostics

```
header: 
  seq: 28
  stamp: 
    secs: 23
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Received packets:"
        value: "34986"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "94"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "2398175"
      - 
        key: "Tx total bytes:"
        value: "790924"
      - 
        key: "Rx speed:"
        value: "21762.000000"
      - 
        key: "Tx speed:"
        value: "537.000000"
  - 
    level: 0
    name: "mavros: GPS"
    message: "3D fix"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Satellites visible"
        value: "10"
      - 
        key: "Fix type"
        value: "3"
      - 
        key: "EPH (m)"
        value: "0.00"
      - 
        key: "EPV (m)"
        value: "0.00"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "3592"
      - 
        key: "Frequency (Hz)"
        value: "1.000000"
      - 
        key: "Vehicle type"
        value: "Quadrotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "MANUAL"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 2
    name: "mavros: System"
    message: "Sensor health"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Sensor present"
        value: "0x00000020"
      - 
        key: "Sensor enabled"
        value: "0x0020000B"
      - 
        key: "Sensor health"
        value: "0x00000020"
      - 
        key: "3D gyro"
        value: "Fail"
      - 
        key: "3D accelerometer"
        value: "Fail"
      - 
        key: "absolute pressure"
        value: "Fail"
      - 
        key: "AHRS subsystem health"
        value: "Fail"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Voltage"
        value: "12.15"
      - 
        key: "Current"
        value: "-1.0"
      - 
        key: "Remaining"
        value: "100.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://:14540@localhost:14557"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "229"
      - 
        key: "Frequency (Hz)"
        value: "10.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "22.892000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.008000000"
---
```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5306 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 24, 30, 31, 32, 36, 74, 331, 76, 77, 83, 340, 230, 105, 111, 241, 245
```

vooon, 2020-08-05:
You can use access through params followed by push.
Other possible fix - check type of param_value and set 0 or 0.0 depending on it.

marcelino-pensa, 2020-08-05:
@vooon , I am not sure if I follow your answer. First of all, what do you mean by "access through params followed by push"?

About your second suggestion, this is the message definition for [ParamValue](http://docs.ros.org/api/mavros_msgs/html/msg/ParamValue.html), which is used by [ParamSet](http://docs.ros.org/api/mavros_msgs/html/srv/ParamSet.html):
```
# Parameter value storage type.
#
# Integer and float fields:
#
# if integer != 0: it is integer value
# else if real != 0.0: it is float value
# else: it is zero.

int64 integer
float64 real
```

The problem is in the last `else` statement, which can either be `integer zero`, or `float zero`. If we send the wrong type to px4, it will complain. If we are trying to set an `integer` variable, we need to send `0`. If we are trying to set a `float` variable, we need to send `0.0`.
If we don't want to change anything on the message definitions of `ParamValue` or `ParamSet`, we would need to find a way to determine the variable type from `param_id` in [ParamSet](http://docs.ros.org/api/mavros_msgs/html/srv/ParamSet.html). Is this possible?

vooon, 2020-08-06:
I mean that you can use regular ROS params manipulation, then call plugin to push changes.
Because xml rpc value keeps the type of variable, it should be fine to differentiate between integer and float.

And because plugin keeps list of all parameters it's also possible to check previous type in set_cb and use proper type.

marcelino-pensa, 2020-08-28:
@vooon , thank you for your feedback!
I finally got some free time to tackle this issue. A solution is proposed in: https://github.com/mavlink/mavros/pull/1474

closed 2020-08-31 by vooon