mavlink/mavros #1469: CONNECTED FAILED

DRAEYE, 2020-08-10:
# Issue details

Hi, i want use mavros to read the data like /mavros/imu/data..., but I came across a problem 

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

### Node logs
when i run the command 
roslaunch mavros px4.launch fcu_url:="udp://:**14555**@192.168.31.214:**14555**",
e216@e216:~$ rostopic echo /mavros/state
header: 
  seq: 4
  stamp: 
    secs: 1597027736
    nsecs: 951196194
  frame_id: ''
connected: True
armed: True
guided: False
mode: "MODE(0xC0)"
system_status: 4
e216@e216:~$ roslaunch mavros px4.launch  fcu_url:="udp://:14555@127.0.0.1:14555"
... logging to /home/e216/.ros/log/2817c1fe-da39-11ea-b96b-d6aa8a9c92b1/roslaunch-e216-28061.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://e216:34698/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
[... 607 lines cut ...]
        value: "0.496583"
      - 
        key: "Last remote time (s)"
        value: "1596975000.627461195"
      - 
        key: "Estimated time offset (s)"
        value: "9061815582.886157990"
```
the connected in/mav/state is false

### Check ID

```
rosrun mavros checkid with roslaunch mavros px4.launch fcu_url:="udp://:**14555**@192.168.31.214:**14555**",
ERROR. I got 1 addresses, but not your target 1:1
Received 343 messages, from 1 addresses
sys:comp   list of messages
  1:240   0, 2, 43, 76, 111, 21

rosrun mavros checkid with roslaunch mavros px4.launch fcu_url:="udp://:**14550**@192.168.31.214:**14555**",
ERROR. I got 0 addresses, but not your target 1:1
Received 0 messages, from 0 addresses
sys:comp   list of messages
```
How can I deal with this problem?

vooon, 2020-08-10:
Hello,

You simply made a loop by setting same port for incoming data (bind) and remote address.

DRAEYE, 2020-08-10:
> Hello,
> 
> You simply made a loop by setting same port for incoming data (bind) and remote address.

Thanks for your reply, I think I haven't understand your words, should i need to finish other necessary settings? I tried the roslaunch mavros px4.launch fcu_url:="udp://:14550@192.168.31.214:14555", but it failed

vooon, 2020-08-11:
Hello,

I mean that you should carefully check port numbers and addresses where you are trying to connect.
Because with `udp://:14550@127.0.0.1:14550` (or any other ip of the host) you connect mavros to itself, so it only hears it's own heartbeats.

Please check that your remote really uses port 14555, and firewall is not dropping packets.
Maybe better to try with TCP first, and when you'll manage to work in that mode, you can switch to UDP.

lukelu0520, 2020-11-12:
@vooon thanks for your tutorial.  We are also very confused by the description of [udp connection in readme](https://github.com/mavlink/mavros/blob/master/mavros/README.md#connection-url), it says **Connection defined by URL, you can use any supported type for FCU**:
udp://[bind_host][:port]@[remote_host[:port]]

What is "bind_host" and "remote_host" ? Why fcu_url have two hosts address here?

Hope to hear your expertise.

Best,
-- Luke

vooon, 2020-11-13:
@lukelu0520 `bind_host` is where mavros will bind(), please read https://en.wikipedia.org/wiki/Berkeley_sockets#bind (in UDP you don't have to listen()).

So `remote_host` is your remote target, usually GCS. Note that any packet that comes to `bind_host` will change your `remote_host:port` automatically.
That's because GCS usually sends packets on it's own, so usage of that address is a kind of auto-detection.

closed 2020-11-18 by DRAEYE

21coolkarni, 2024-01-08:
I am having same issue on  a serial port