mavlink/mavros #1471: include could not find load file: MavrosMavlink and EnableCXX11

Cristian-wp, 2020-08-25:
### Issue details

Hi, for a project we have a repository in Azure DevOps and we are trying do create a common repository to work.
The structure of the repository is the following one:

(root project folder)
       catkin_ws (catkin workspace folder)
              build
             devel
             src
                  Firmware (repository PX4/Firmware)
                  avoidance (repository PX4/avoidance)
                  sei-ros-beta (repository)
                               mavlink
                               mavros
                               sample-package
                              ... another package
                              ... another repo

When I try to compile the cloned repository I get this errors:
````
Errors     << libmavconn:cmake /home/sei/AzureDevOps/SEI/catkin_ws/logs/libmavconn/build.cmake.000.log
CMake Error at /home/sei/AzureDevOps/SEI/catkin_ws/src/sei-ros-beta/mavros/libmavconn/CMakeLists.txt:15 (include):
  include could not find load file:

    EnableCXX11

CMake Error at /home/sei/AzureDevOps/SEI/catkin_ws/src/sei-ros-beta/mavros/libmavconn/CMakeLists.txt:16 (include):
  include could not find load file:

    MavrosMavlink
````
I had already check if there is any missing dependency with 
````
sei@sei:~/AzureDevOps/SEI/catkin_ws$ rosdep install --from-paths src --ignore-src
#All required rosdeps installed successfully
````
But as you can see all seems ok.
Can you please help me to solve this problem?

### MAVROS version and platform

Mavros: Release 1.3.0
ROS: Melodic
Ubuntu: 18.04 LTS

### Autopilot type and version

[  ] ArduPilot
[ x  ] PX4

Version: 1.10.1 (stable)

### Output of catkin build -c 

````
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
------------------------------------------------------------------------
[build] Found '12' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 1.4 seconds ]             
Starting  >>> mavlink                                                          
Starting  >>> mavros_msgs                                                      
Starting  >>> sample-package                                                   
Finished  <<< sample-package                       [ 2.8 seconds ]             
Finished  <<< mavlink                              [ 3.9 seconds ]             
Starting  >>> libmavconn                                                       
[... 164 lines cut ...]
    interpreter.execute(self.code, locals)
  File "/usr/bin/empy", line 2595, in execute
    _exec(statements, self.globals, locals)
  File "/usr/bin/empy", line 42, in _exec
    exec("""exec code in globals""")
  File "<string>", line 1, in <module>
  File "<string>", line 1, in <module>
  File "<string>", line 16, in <module>
ValueError: common dialect not listed!
[31m[1mCMake Error[0m at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):

  execute_process(/home/sei/AzureDevOps/SEI/catkin_ws/build/libmavconn/catkin_generated/env_cached.sh
  "/usr/bin/python2" "/usr/bin/empy" "--raw-errors" "-F"
  "/home/sei/AzureDevOps/SEI/catkin_ws/build/libmavconn/catkin_generated/mavlink.context.py"
  "-o"
  "/home/sei/AzureDevOps/SEI/catkin_ws/build/libmavconn/catkin_generated/include/mavconn/mavlink_dialect.h"
  "/home/sei/AzureDevOps/SEI/catkin_ws/src/sei-ros-beta/mavros/libmavconn/include/mavconn/mavlink_dialect.h.em")
  returned error code 1
[36mCall Stack (most recent call first):[0m
  /opt/ros/melodic/share/catkin/cmake/em_expand.cmake:25 (safe_execute_process)
  CMakeLists.txt:58 (em_expand)

[36m--[0m Configuring incomplete, errors occurred!

````

vooon, 2020-08-25:
Somehting wrong with cmake config. That modules here: https://github.com/mavlink/mavros/tree/master/libmavconn/cmake/Modules

Cristian-wp, 2020-08-31:
Thank you @vooon  for the reply, do you know how to fix it?
I have also try this two ways:
1) Use the 1.2 version and the error disappear
2) Reinstall the system, but the error persist

How can I correctly modify the cmake config?

closed 2026-11-07 by github-actions[bot]