mavlink/mavros #1472: [Bug][Test video] Yaw data is not decreasing when I turn left from 0, but suddenly changes to 3.14

cnpcshangbo, 2020-08-26:
----

### Issue details

Hello, I am following [this setup](https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-2-complete-installation-and-indoor-non-gps-flights/43405). However, I noticed that the yaw data is not decreasing when I turn left from 0, but suddenly changes to 3.14. Here's my [video](https://youtu.be/fGTIb4XI0IU). Both the Terminal and Rivz are showing rostopic of /mavros/vision_pose/pose. The Mission Planner is showing the output of Mavros. I have tried Mavros 1.0, 1.1, 1.2, and the latest on GitHub. They have the same result. Did I miss something? Thanks.

### MAVROS version and platform

Mavros: 1.0, 1.1, 1.2
ROS: M
Ubuntu: 18.04

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: 4.1.0-dev

### Node logs

```

root@nvidia-desktop:/home/nvidia# roslaunch mavros apm.launch
... logging to /root/.ros/log/67d5b5b2-e7bd-11ea-8e30-00044bf20a08/roslaunch-nvidia-desktop-8520.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia-desktop:40421/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
[... 435 lines cut ...]
        key: "Last RTT (ms)"
        value: "2.148059"
      -
        key: "Mean RTT (ms)"
        value: "2.979962"
      -
        key: "Last remote time (s)"
        value: "565.797229000"
      -
        key: "Estimated time offset (s)"
        value: "1598461117.410373688"
---
```

### Check ID

```
➜  ~ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 1611 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 150, 129, 137, 152, 147, 22, 24, 27, 29, 30, 193, 33, 35, 36, 165, 42, 178, 62, 65, 74, 77, 163, 111, 241, 116, 125
```

closed 2020-09-03 by amilcarlucas

cnpcshangbo, 2020-09-03:
Hi @amilcarlucas, Thanks for your answer. It is in radians. I think what we need is when turning left, the output goes from 0 to negative until -pi. Do you think that is corresponding to the yaw definition of Ardupilot?

amilcarlucas, 2020-09-04:
It uses normal aeronautical convention, all positive in the interval [0, 2pi[

cnpcshangbo, 2020-09-06:
If I want to modify this, do you have some ideas where should I change it?
Thanks,
Bo