mavlink/mavros #1473: Modify Geofence via Mavros Interface?

moooeeeep, 2020-08-28:
### Issue details

I would like to specify a geofence for my drone from its onboard companion computer. There I have a Python node, that takes care of the mission control and interfaces to the FCU via mavros. I would like to specify a geofence from within that node during flight, based on sensor data that is processed on the companion.

In the docs I didn't find a plugin for this, mavros_msgs doesn't seem to contain the relevant messages and services, so I assume that this functionality does not yet exist.

I only found this related feature request: https://github.com/mavlink/mavros/issues/1157 

Has anyone dealt with this before (and can share any workarounds)? Did anyone make progress implementing this?

Dan-Moss, 2020-09-24:
I would also like to know if this is possible.

zlowell2, 2020-12-16:
Any progress made on this?

Jaeyoung-Lim, 2020-12-16:
@moooeeeep @Dan-Moss @zlowell2 Since the command plugin provides an interface to send `MAV_CMD` messages, you can set the geofence using this interface

zlowell2, 2020-12-16:
@Jaeyoung-Lim which .msg file would be used to convey this interface though. The CommandCode.msg file contains the command type so we can reference/define the command in this message as something like:

command = NAV_FENCE_POLYGON_VERTEX_INCLUSION = 5001

but where do we define the longitude, latitude, alt, etc. associated with the geofence and which .msg file would this be contained in, before we send the message to the command node?

JqkerN, 2021-01-20:
Would also know how to do that, did you figure out a way of sending geofence polygons through mavros?

moooeeeep, 2021-01-20:
@Jaeyoung-Lim  I feel this is more complicated than that. Geofences are not specified in a command message. You have to transmit them via the [mission protocol](https://mavlink.io/en/services/mission.html#uploading_mission). The mavros mission plugin, which would seem relevant here, doesn't support changing the mission type to MAV_MISSION_TYPE_FENCE. Probably one could copy and paste most of the code and create a specific fence plugin though. But I felt this would go beyond the effort I was able to put in at that point. 

Eventually, I ended up specifying the geofence via an additional pymavlink interface that I connected locally via the gcs_url that I configured mavros to open up for this. I consider this a hacky workaround and I'm not exactly proud of getting it work this way, but at the end of the day it enabled me to transmit a single fence to the FCU. Which was enough to cover my use case. I didn't manage to transmit multiple fences, as the next one would always overwrite the previously transmitted fence. Not sure if the bottleneck was my lack of knowledge of this whole stack or again a limitation of the interface I used.

closed 2026-11-07 by github-actions[bot]