mavlink/mavros #1475: mavros offboard mode can arm but can't take off

haocou, 2020-08-29:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hello, I am using mavros offboard mode(official code: https://dev.px4.io/v1.10/en/ros/mavros_offboard.html) to control the UAV. But, the UAV can’t work as expected. The arm first actuated and then disarmed. This happened repeatedly. I have already switched to the offboard mode, It can’t take off. I have checked the hardware. And the mocap message is remapping from /rigidbody_1/pose to /mavros/vision_pose/pose. The frame between mocap and UAV are checked. I don’t know what happened.

Hardware: raspberry pi, and telem2 port to raspberry pi(the connection has been checked).

The uploaded logs are below(corresponding to px4.launch logs, five logs)
1)
https://logs.px4.io/plot_app?log=cccb4439-41d3-4a2f-99d1-82c6b5ffec6c
2)
https://logs.px4.io/plot_app?log=c9708e5f-1972-47f3-9ab2-aaef9f83ecd1
3)
https://logs.px4.io/plot_app?log=cac1201f-808c-4af7-9053-e826a2522db3\
4)
https://logs.px4.io/plot_app?log=720049f9-93e1-4fa6-81a3-76b2961a3822
5)
https://logs.px4.io/plot_app?log=61c320be-ec6f-4098-bdea-c67ba68369b6

I would appreciated if you can help me.

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

PX4
Version: 1.10.1

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```
... logging to /home/bitcq/.ros/log/416eb4bc-ea17-11ea-bcc1-30b49ea1a4df/roslaunch-bitcq-desktop-1731.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.2.77:42097/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
[... 449 lines cut ...]
      - 
        key: "Last RTT (ms)"
        value: "3.448906"
      - 
        key: "Mean RTT (ms)"
        value: "4.845694"
      - 
        key: "Last remote time (s)"
        value: "85.278020000"
      - 
        key: "Estimated time offset (s)"
        value: "1598583455.910658360"
---

### Check ID

```
rosrun mavros checkid
```
OK. I got messages from 1:1.

---
Received 5222 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 140, 141, 147, 30, 31, 32, 36, 74, 331, 76, 77, 83, 340, 230, 105, 111, 241, 245

DogukanAltay, 2021-01-21:
Can you check couple of things?

1. Have you set EKF2_AID_MASK to 24 (vision pose, vision yaw checkboxes)
2. Have you set EKF2_HGT_MODE to vision?
3. Are you sure that your positioning data has the proper header time?

I have had the same issue as you. After I have did the 3rd my issue resolved.

AllenHuangDongyue, 2021-01-22:
> Can you check couple of things?
> 
> 1. Have you set EKF2_AID_MASK to 24 (vision pose, vision yaw checkboxes)
> 2. Have you set EKF2_HGT_MODE to vision?
> 3. Are you sure that your positioning data has the proper header time?
> 
> I have had the same issue as you. After I have did the 3rd my issue resolved.

How can I check whether my positioning data has the proper header time? I used the official code.

DogukanAltay, 2021-01-26:
If you are using rospy, below code snipped will yield a correct positioning data with proper header time:

```
message = PoseStamped()
message.header.stamp = rospy.Time.now()
message.pose.position.x = x  # any dummy value or your own positioning data
message.pose.position.y = y  # any dummy value or your own positioning data
message.pose.position.z = z  # any dummy value or your own positioning data
```

closed 2026-11-06 by github-actions[bot]