mavlink/mavros #1478: Error bad_weak_ptr when using roslaunch after success build from source

tinng81, 2020-09-09:
### Issue details

After success [source installation](https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation) of *mavros* and dependencies, I encountered error `bad_weak_ptr` when using [launch instruction](https://dev.px4.io/v1.9.0/en/simulation/ros_interface.html#launching-rossimulation) from PX4.

`roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`

Some other steps I have tried:
- `catkin clean` and re-build all packages.
    - Build success, same error

- Remove everything, re-install dependencies and build from [source installation](https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation) again.
    - Build success, same error

- Instead of using `wstool update`, cloned both *mavlink* and *mavros* as submodules and update with `git submodule update --init --recursive`.
    - Build failed, force update *mavlink* with `git submodule update --remote -f mavlink`; afterwards, build success but still same error

### MAVROS version and platform

Mavros: 1.30.0
ROS: Melodic
Ubuntu: Bionic 18.04

### Autopilot type and version

[  ] ArduPilot
- [x] PX4

Version: 1.11.0

### Node logs

```
NODES
  /
    mavros (mavros/mavros_node)

auto-starting new master
process[master]: started with pid [24300]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to af45f612-f26e-11ea-ad5e-001c42db7e93
process[rosout-1]: started with pid [24311]
started core service [/rosout]
process[mavros-2]: started with pid [24319]
[ INFO] [1599636780.251461931]: FCU URL: udp://:14540@127.0.0.1:14557
[ INFO] [1599636780.261582909]: udp0: Bind address: 0.0.0.0:14540
[ INFO] [1599636780.261684315]: udp0: Remote address: 127.0.0.1:14557
[ INFO] [1599636780.263160901]: GCS bridge disabled
terminate called after throwing an instance of 'std::bad_weak_ptr'
  what():  bad_weak_ptr
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 24319, exit code -6, cmd /home/winng/Documents/Belvedere/catkin_ws/devel/lib/mavros/mavros_node __name:=mavros __log:=/home/winng/.ros/log/af45f612-f26e-11ea-ad5e-001c42db7e93/mavros-2.log].
log file: /home/winng/.ros/log/af45f612-f26e-11ea-ad5e-001c42db7e93/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

### Diagnostics

```
rostopic echo -n1 /diagnostics
N/A
```

[... 117 lines cut ...]
[roslaunch][INFO] 2020-09-09 14:33:00,384: [mavros-2] killing on exit
[roslaunch.pmon][INFO] 2020-09-09 14:33:00,386: ProcessMonitor exit: killing rosout-1
[roslaunch][INFO] 2020-09-09 14:33:00,386: [rosout-1] killing on exit
[roslaunch][INFO] 2020-09-09 14:33:00,386: process[rosout-1]: killing os process with pid[24311] pgid[24311]
[roslaunch][INFO] 2020-09-09 14:33:00,386: [rosout-1] sending SIGINT to pgid [24311]
[roslaunch][INFO] 2020-09-09 14:33:00,386: [rosout-1] sent SIGINT to pgid [24311]
[roslaunch][INFO] 2020-09-09 14:33:00,486: process[rosout-1]: SIGINT killed with return value 0
[roslaunch.pmon][INFO] 2020-09-09 14:33:00,487: ProcessMonitor exit: killing master
[roslaunch][INFO] 2020-09-09 14:33:00,487: [master] killing on exit
[roslaunch][INFO] 2020-09-09 14:33:00,487: process[master]: killing os process with pid[24300] pgid[24300]
[roslaunch][INFO] 2020-09-09 14:33:00,487: [master] sending SIGINT to pgid [24300]
[roslaunch][INFO] 2020-09-09 14:33:00,487: [master] sent SIGINT to pgid [24300]
[roslaunch][INFO] 2020-09-09 14:33:00,688: process[master]: SIGINT killed with return value 0
[roslaunch.pmon][INFO] 2020-09-09 14:33:00,688: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2020-09-09 14:33:00,688: ProcessMonitor exit: pmon has shutdown
[roslaunch][INFO] 2020-09-09 14:33:00,743: process monitor is done spinning, initiating full shutdown
[roslaunch][INFO] 2020-09-09 14:33:00,743: runner.stop()
[roslaunch][INFO] 2020-09-09 14:33:00,743: shutting down processing monitor...
[roslaunch][INFO] 2020-09-09 14:33:00,743: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 139721410692864)>
[roslaunch.pmon][INFO] 2020-09-09 14:33:00,744: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139721410692864)>
[roslaunch][INFO] 2020-09-09 14:33:00,744: ... shutting down processing monitor complete
[roslaunch][INFO] 2020-09-09 14:33:00,744: done
[roslaunch.pmon][INFO] 2020-09-09 14:33:00,744: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139721410692864)>
[rospy.core][INFO] 2020-09-09 14:33:00,745: signal_shutdown [atexit]
```

marcelino-pensa, 2020-10-14:
Is this happening every time you launch `mavros`?
I've been seeing this happening in multiple devices, but it doesn't happen every time, only occasionally

tinng81, 2020-10-20:
@marcelino-pensa The issue is indeed persistent, I have resolved to using the prebuilt binary for the moment.

vooon, 2021-11-26:
Should be fixed on upcoming 1.12.0.

closed 2021-11-26 by vooon